arduino/ArduinoCore-nRF528x-mbedos #3: Rename to ArduinoCore-mbedos

PTS93, 2019-08-01:
If I understand the intention correctly this is a generic library to bring the Arduino API into Mbed. In that case, maybe it make sense to reflect hat in the name to avoid confusion?

This is a fantastic development btw!

pnndra, 2019-08-01:
sort of. mbed supports various platforms and this one is tested only on nRF528x. likely we'll be moving to a common core for several targets but this is not the case for now. will be nice to see comments and contributions to make this grow.

PTS93, 2019-08-01:
Ah yea now that I had a closer look it makes more sense. So this is an Arduino core definition with all the meta data for the Arduino IDE but also includes all the code for the translation between mbed and arduino API.

Sounds like a great idea to seperate that in the near future so other potential cores that wish to implement a different mbed compatible target can just specify the translation layer and toolchain as a dependency instead of creating a copy of this.

lxrobotics, 2019-09-16:
Closing this issue since not directly related to `ArduinoCore-nRF528x-mbedos`. Thank you for your input which has been duly noted - we might move to a common core in the future but for now `ArduinoCore-nRF528x-mbedos` will stay `ArduinoCore-nRF528x-mbedos`.

closed 2019-09-16 by lxrobotics