arduino/ArduinoCore-nRF528x-mbedos #5: [SPI] missing  setClockDivider, setBitOrder, and setDataMode API's

sandeepmistry, 2019-08-19:
Originally reported by @LittleHobbyShop here: https://github.com/adafruit/Adafruit-GFX-Library/issues/242

@facchinm any thoughts on this?

facchinm, 2019-08-19:
Indeed, setClockDivider(), setBitOrder(), and setDataMode() were marked as deprecated since the introduction of SPISettings and thus not included in https://github.com/arduino/ArduinoCore-API . I'm a bit concerned in reintroducing them and I'd prefer if GFX library could be patched.
The patch shouldn't hurt any core since everyone already implements SPISettings

facchinm, 2019-08-19:
Nevermind, I found the issue; SPI.h is missing `SPI_HAS_TRANSACTION` define. The library compiles fine after defining it; I'm going to add it to API repo (since transactions support is compulsory) and publish an updated core when done :wink:

facchinm, 2019-08-19:
Fixed by https://github.com/arduino/ArduinoCore-API/commit/862fb23829fe333e8218934cd3437ac16246231f and tagged core 1.1.1

closed 2019-08-19 by facchinm

ladyada, 2019-08-19:
thanks! that was easy :)