arduino/ArduinoCore-nRF528x-mbedos #11: analogReference(uint8_t mode) function not implemented

fgaetani, 2019-09-20:
Unfortunately, I noticed that the analogReference(uint8_t mode) function is not implemented in wiring_analog.cpp. It would be useful to have a function that uses the internal voltage reference at 0.6V and the relative multiplier. I await updates.

facchinm, 2019-09-23:
Hi @fgaetani ,
thanks for spotting this, I'll be working on a fix soon :wink:

fgaetani, 2020-03-23:
Hi @facchinm, are there any news regarding this implementation? Thank you!

facchinm, 2020-03-23:
No news unfortunately, mbed code still configures the channel on init() (https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/analogin_api.c#L96) so overriding it is impossible ATM. 
Anyway it could be a target of https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/31 if I find some time to complete it.

fgaetani, 2020-03-23:
Ok thank you very much, I hope soon!

Gerriko, 2020-05-15:
Any update on this?

I quite like Adafruit's implementation although no idea how they've done it.

https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/nrf52-adc

facchinm, 2020-06-22:
`analogReference` is now implemented for nRF52 microcontrollers on https://github.com/arduino/ArduinoCore-mbed (beta version, can be tested by downloading Portenta core from Board manager) or [beta branch](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/81).
Testing and reporting results is, as always, appreciated :slightly_smiling_face:

closed 2020-07-30 by facchinm