arduino/ArduinoCore-nRF528x-mbedos #12: Ethernet library doesn't compile because of the namespace arduino::

jandrassy, 2019-09-21:
Ethernet library doesn't compile with this core because the IPAddress class in this core's release is in name space `arduino`. there are two problems:
1) the Ethernet library friend classes are not friended because they are at global scope, not in namespace arduino
2) the INADDR_NONE doesn't match the extern declaration. extern is in arduino::, the definition is not

fix 1 in IPAddress.h. forward declare classes before `namespace arduino {`
```
class EthernetClass;
class DhcpClass;
class DNSClient;
```
declare friends as
```
    friend ::EthernetClass;
    friend ::DhcpClass;
    friend ::DNSClient;
```
(or fix the Ethernet library, so it doesn't use private members of the IPAddress class)

fix 2 in IPAddress.cpp
```
const IPAddress arduino::INADDR_NONE(0,0,0,0);
```

facchinm, 2019-09-23:
Hi @jandrassy ,
thank you very much for the fixes! Could you open a PR on https://github.com/arduino/ArduinoCore-API/tree/namespace_arduino (the `namespace_arduino` branch is the important part here ) so we can keep track of the contributions?
Thanks again.

jandrassy, 2019-09-23:
> 
> 
> Hi @jandrassy ,
> thank you very much for the fixes! Could you open a PR on https://github.com/arduino/ArduinoCore-API/tree/namespace_arduino (the `namespace_arduino` branch is the important part here ) so we can keep track of the contributions?
> Thanks again.

yes, of course, I could not find it. I didn't look at the branches. I would fix the Ethernet library too, but I think there is a policy of not changing the established code.

facchinm, 2019-09-23:
It's probably better not to touch Ethernet library before we transition most of the cores :slightly_smiling_face:

closed 2019-09-27 by jandrassy