arduino/ArduinoCore-nRF528x-mbedos #13: Compiling Issue for use of Servo.h Library

andamado997, 2019-09-23:
I was trying to make a test sketch to control a servo using my new Nano 33 Sense BLE board, I then received this error when I tried to compile the program:

 WARNING: library Servo claims to run on (avr, megaavr, sam, samd, nrf52, stm32f4) architecture(s) and may be incompatible with your current board which runs on (mbed) architecture(s).
In file included from C:\Users\DellPC\AppData\Local\Temp\Temp1_Arduino Board Code.zip\ServoBody\ServoBody.ino:1:0:

C:\Users\DellPC\Documents\Arduino\libraries\Servo\src/Servo.h:75:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

  ^~~~~

Multiple libraries were found for "Servo.h"
 Used: C:\Users\DellPC\Documents\Arduino\libraries\Servo
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\Servo
exit status 1
Error compiling for board Arduino Nano 33 BLE.

Can I expect this issue to resolve this in the very near future? I can understand that its a fairly new piece of hardware but I wasn't aware that there was no current support for the Servo.h Library, which I need to use for an upcoming project. I'd hate to have to return and exchange for another Arduino board since I like the variety of sensors loaded on the Nano 33 Sense.

facchinm, 2019-09-23:
@andamado997 the porting will be ready very soon, don't worry :wink: I'll keep you posted as soon as there's a pull request ready so you can test it before merge.

andamado997, 2019-09-24:
Sounds Good! Keep us posted.

facchinm, 2019-09-24:
Hi @andamado997 ,
I opened the pull request here https://github.com/arduino-libraries/Servo/pull/32
If you could test it and report back ? Thanks!

andamado997, 2019-09-24:
Just did a test compile on my IDE and it seems like that did it. I'll try a more involved test with the board and a motor later today to make double sure. Thanks again!

Netguykb, 2019-12-16:
Has there been a resolution to a NEW servo library that will work with Nano 33 BLE??

facchinm, 2019-12-17:
Hi @Netguykb ,
the code was merged but the lib was not released, my bad. Just did it now, it should be available on library manager in an hour more or less.

closed 2019-12-17 by facchinm

Netguykb, 2019-12-17:
@facchinm Thanks for the update. I assume the library you're referring to is the Arduino Library Manager. If so whats the name I can use to search for this update? Thanks again!

facchinm, 2019-12-17:
Simply `Servo` :slightly_smiling_face:

Netguykb, 2019-12-17:
> Simply `Servo` 🙂

Ok, But there 47 of them with the name 'Servo'

The two I have loaded are:
EduIntro by Arduino LLC Version 0.0.12
Servo by Michael Margolis, Arduino Version 1.1.6

Are any of these two yours? Thanks again!

per1234, 2019-12-17:
>Servo by Michael Margolis, Arduino Version 1.1.6

That's the one.