arduino/ArduinoCore-nRF528x-mbedos #17: Compatibility: yield() weak binding, delay() calling yield()

dok-net, 2019-09-25:
In official AVR Arduino core, delay() is based on yield(), which is turn has weak binding and is overridable.
I am asking that yield() has a weak binding to __yield(), which shall contain the default implementation for the given platfrom, and that delay() on any Arduino platform must call yield() at least once, preferably at the end of delay(), for compatibility with libraries like Scheduler and similar implementations.

dok-net, 2019-09-27:
PR #18 implements this scheme.

facchinm, 2020-03-11:
This core is based on an actual RTOS, so I don't see any reason to allow the override.
Of course a library implementing Scheduler APIs can be created but it will be based on actual threads.

closed 2020-03-11 by facchinm

dok-net, 2020-03-11:
@facchinm I obviously disagree, and shouldn't be up to the Arduino core of any given platform to force something on developers that they should have the freedom to decide. Because, in fact, the simple change here makes no difference at all if not exploited. Ever being able to shim yield() and delay() may be precondition for any sort of library, many of those may not need porting to this platform, except now they do, because __yield() and __delay() don't exist.
Please reconsider?