arduino/ArduinoCore-nRF528x-mbedos #19: Use MBED with Arduino MBED compatible board ?

charly37, 2019-09-30:
Hi,

Since the nano 33 are based on MBED and run MBEDos (with arduino core on top of it) I wonder if it is possible to use MBED with arduino core on such board ? How would you do that (which target to use and how to connect the board to the PC to upload ?)

Thx

facchinm, 2019-10-01:
Hi @charly37 ,
this is a very well made guide to achieve this http://blog.janjongboom.com/2019/08/01/arduino-mbed.html . 
Upload is not integrated since we don't use "Mass storage" upload but a serial based one (it could, however, be a future improvement). 
To upload out of Arduino IDE you'll need a copy of `bossac` uploader (based on this branch https://github.com/arduino/BOSSA) with the following commandline

`bossac --port={serial.port} -U -i -e -w "{build.project_name}.bin" -R`

closed 2019-10-01 by facchinm

charly37, 2019-10-01:
Hi @facchinm and Thx for your help.
I found the blog entry you provide when i was doing research but the upload part was not very clear to me so I open a ticket on their github too > https://github.com/janjongboom/arduino-mbed-mashup/issues/2

I was trying to understand their advice for upload "you just need to flash through the Arduino IDE" but I think your solution may be easier with BOSSAC. 
Nevertheless I have more of a philosophical question about why MBED don t support the board directly...since the Arduino guys made the efforts to have MBED running on the nano 33 it would make sense to have it recognize as a MBED compatible board. I think it may widen the possible audience of people that may buy this board. 

Thx again. I will try BOSSAC as soon as possible

facchinm, 2019-10-01:
The upload problem is related with the way mbed-compatible boards work. They normally ship with a full fledged debugger (based on [CMISIS-DAP](https://arm-software.github.io/CMSIS_5/DAP/html/index.html)) that exposes a mass storage interface (like a USB stick) so the mbed uploader simply copies the binary file in there and this triggers the upload.
Adding an upload method is a bit tricky (feasible, but tricky).
If you use plain mbed with ARDUINO_NANO33BLE target (from mbed 5.14.0 onwards) you'll get an image that flashes at 0x0, overwriting the bootloader.
if you want to keep the original bootloader you'll need to relocate the image at 0x10000 (see https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/mbed-os-to-arduino#L62 )

charly37, 2019-10-02:
Thx again @facchinm for all the info. I was unable to use the target "ARDUINO_NANO33BLE" because i was on mbed 5.13.1 but i now have it after i upgrade to 5.14 ;)
I did not had time to try the upload with BOSSAC yet (I tried another solution with the arduino IDE - more like a hack - https://os.mbed.com/questions/87137/Code-with-Mbed-studio-for-Arduino-nano-3/?c=33946). Just sharing in case someone else in the future face similar issue/topic ;)
I will try BOSSAC ASAP and let you know (may get some issue to have that working on windows i guess).
Thx again for you explanation....I use to do simple project and stick with the Arduino IDE and never learn much about bootloader and uploader so your info help me better understand who it s working behind (especially with these new board and mbedOs).
**If you have some time** I have a question about the "CMISI-DAP"... I guess it s what I saw on some board called "DAP-link" like "The development kit comes with a fully integrated debugger (also known as DAPLink) that provides USB drag-and-drop programming, USB Virtual COM port and CMSIS-DAP interface." on https://www.seeedstudio.com/nRF52840-Micro-Development-Kit-p-3079.html
I m curious to understand if this is a program running on the chip on the board (like a bootloader) and part of MBED OS or maybe more like another dedicated hardware/chip on the board that act as a hardware programmer between the PC and the ARM chip ?

facchinm, 2019-10-02:
@charly37 glad it helped :slightly_smiling_face: 

About CMSIS-DAP, it's usually implemented as a dedicated chip (can be as small as an LPC11U35) and the sources to build your own are freely available here https://github.com/ARMmbed/DAPLink . 
The on-chip solution is usually impractical since it would need some "coprocessor" that doesn't get stopped when debugging the "main" chip, but the other functionalities (drag and drop, serial) can be implemented in a bootloader with not much effort :wink:

charly37, 2019-10-05:
I made some progress and burn my Nano 33 BLE bootloader in the mean time ;) (I manage to recover with a J-Link EDU in case it happen to anybody else I share my notes http://djynet.net/?p=969)

some context for the command 
ArduinoBlink2.ino.bin > code compile in Arduino IDE
MbedBlink3.bin > code compile in MBED Studio

So... I manage to upload with BOSSAC a program written by Arduino (basically the same command that Arduino IDE is doing) with:
 `.\bossac.exe -d --port=COM5 -o 0x0000 -U -i -e -w 'C:\Build\ArduinoBlink2.ino.bin' -R`
(I upload at address 0x0...works too without this args like the IDE is doing)
Then I tried to upload my MBED program with BOSSAC too at address 0x10000 as suggested
`.\bossac.exe -d --port=COM5 -o 0x10000 -U -i -e -w 'C:\Build\MbedBlink3.bin' -R`
but it do not works ;(
Then always with BOSSAC I tried to upload my MBED program to address 0x0 and it do not works either.

Then since I received my J-Link EDU in the meantime I decided to try to use J-Flasher to upload and
I can upload my program MbedBlink3.bin at address 0x0 and it works (tha's how i now my code is OK)
I can not upload my Arduino program at address 0x0
I can upload the bootloader at 0x00 (and thus use again the card for BOSSAC upload)

I probably miss a piece of information to be able to understand the results of the test but i m surprise by the fact that uploading the Arduino program with BOSSAC is done at address 0x00....it probably mean that when I compile my arduino program on the IDE it "add" the bootloader in it at adresse 0x00
I m also surprise that uploading the arduino program with J-Flash at address 0x00 did not works since I would had except the same results that what i saw with BOSSAC when uploading at address 0x0 my arduino program.

charly37, 2019-10-11:
So i found the issue is with the upload with bossac.

If i erase the chip, then add the bootloader, then upload with bossac at 0x10000 and check memory I have nothing
[image]

if i erase the chip, then add the bootloader, then upload with j-link at 0x10000 and check memory I have data and everything works fine.
[image]

So the issue is located in the upload. I m just sharing in case it may help people in the future. For now I will just use J-Link directly (the whole goal to go through the bootloader was to avoid buying a costly programmer but now i bought one and j-link edu mini was cheap in fact 15$....)

WilliamGFish, 2019-11-23:
> I made some progress and burn my Nano 33 BLE bootloader in the mean time ;) (I manage to recover with a J-Link EDU in case it happen to anybody else I share my notes http://djynet.net/?p=969)
> 
> some context for the command
> ArduinoBlink2.ino.bin > code compile in Arduino IDE
> MbedBlink3_nooffset.bin > code compile in MBED Studio (without mbed_app_start": "0x10000")
> MbedBlink3_offset.bin > code compile in MBED Studio (with mbed_app_start": "0x10000")
> 
> ### With bossac
> So... I manage to upload with BOSSAC a program written by Arduino (basically the same command that Arduino IDE is doing) with:
> `.\bossac.exe -d --port=COM5 -o 0x0000 -U -i -e -w 'C:\Build\ArduinoBlink2.ino.bin' -R`
> (I upload at address 0x0...works too without this args like the IDE is doing)

I think you'll find that Arduino it but only burns your application it also burns the bootloader. I suggest having a look in the Arduino IDE board files.