arduino/ArduinoCore-nRF528x-mbedos #20: Arduino nano 33 BLE Sense UART is not working properly

gianmarcov, 2019-10-06:
I created a library for uart communications:
https://github.com/gianmarcov/arduino_vesc/tree/develop

With the WROOM ESP32 it's working and i have no problem, with the nano 33 BLE Sense i tried the same sketch changed the Serial2 to Serial1 and nothing happens, i also swapped rx and tx to the other board without results.

I have remote debugged the other controller and i see only 2 bytes incoming the other not.

So i soldered the wires to SWD pins and i flashed my own code without Andruino Core etc. directly using the Nordic nRF52 SDK and now it's working without problems.

So something is going wrong in the Arduino Core or MBed, that all bytes are not transferred via tx.

if you have questions or i should test some code changes don't hesitate to contact me.

Thread that i created in the Forum:
https://forum.arduino.cc/index.php?topic=639277.0

gianmarcov, 2019-10-06:
Hello,

I think i found the Problem:
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/a89067f0cbb9f120a0382295322684b71e6d0243/cores/arduino/Serial.cpp#L76-L81

I call in my library Serial.write(buf, len) and i saw it's not implemented, so if i write every single byte it works.

facchinm, 2019-10-07:
@gianmarcov the problem is not the non implementation (because it gets inherited by Print.write() so it executes the slow byte-by-byte write)  but the return value of `putc()` .
A patch is coming :wink:

facchinm, 2019-10-07:
@gianmarcov can you give a spin to https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/21 and report if it work for you? Thanks!

gianmarcov, 2019-10-07:
@facchinm when i'm home, i will test the changes and report back. grazie mille!

gianmarcov, 2019-10-07:
@facchinm bad news, replaced in the Version 1.1.2 the Serial.ccp and Serial.h with yours, but it doesn´t works.

facchinm, 2019-10-14:
@gianmarcov sorry to hear that. Do you have access to an oscilloscope to check what is going on? From my tests all bytes are being sent (after the patch) and calling `write()` is blocking as per Arduino API specifications, so it should work just as every other core

gianmarcov, 2019-10-16:
@facchinm yesterday i attached my Rigol Oszilloskope and decoded de TX Pin, i have the same result using for-loop write and write(buffer, len). So the patch is working right, the bytes are identical.
The Issue is resolved ;-)

closed 2019-10-21 by facchinm