arduino/ArduinoCore-nRF528x-mbedos #23: Bootloader source code

fgaetani, 2019-10-10:
Hello, I'm working on a project using this board, is it possible to have the source code of the bootloader? Is it a Bossa bootloader, right? I would like to implement a DFU mode like the Nordic bootloader.

Thank you very much!

bradjc, 2019-10-14:
Just chiming in that it would be very useful to be able to have some info on how to compile the bootloader for the Nano 33 BLE Sense.

bradjc, 2019-12-10:
Does reading work on the bootloader that is flashed on the board by default? I'm trying:

```
bossac --port=/dev/cu.usbmodem14201 -U -i -o 0x20000 --read=4096 out.bin -d
```
and getting:
```
Set binary mode
version()=Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Connected at 921600 baud
identifyChip()=nRF52840-QIAA
write(addr=0,size=0x34)
writeWord(addr=0x30,value=0x400)
writeWord(addr=0x20,value=0)
version()=Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Read 4096 bytes from flash
[                              ] 0% (0/1 pages)read(addr=0x20000,size=0x1000)

SAM-BA operation failed
```
Writing is working as expected.

bradjc, 2019-12-13:
@facchinm Sorry to keep bugging on this, but could you leave a quick comment on the state of the bootloader? Is it: closed source, just going to take some time to make available, or already available somewhere?

We want to use the board for an open source project ([tock](https://github.com/tock/tock)), but our use case requires being able to read flash before writing new apps to the board.

rpavlik, 2020-01-06:
I'm also wondering about this - I much prefer the UF2 bootloaders used on a variety of boards, including all recent Adafruit ones (see https://github.com/adafruit/Adafruit_nRF52_Bootloader for their nrf52840 port that includes some nano 33 ble support), but since SAM-BA isn't a terribly common protocol on nrf chips, that's not supported by default with the nrf52 uf2 bootloader so the Arduino IDE breaks. If I had the bootloader source, I could possibly integrate that protocol as well.

bradjc, 2020-05-07:
Can the bootloader update itself?

rpavlik, 2020-05-07:
Not sure, but I couldn't replace it with another bootloader without using a debug tool.

bradjc, 2020-05-07:
I'm now thinking about a trojan horse application that would replace the bootloader on first boot.

rpavlik, 2020-05-07:
Yeah, that's how many of the Adafruit UF2 bootloader upgraders/installers work. I considered it but didn't have the time to dedicate to making it.

sabas1080, 2020-06-26:
Hi everyone

Bossa for nrf52 is here https://github.com/arduino/BOSSA

rpavlik, 2020-06-26:
I can't find the bootloader in that code, though, only the programming utility (which mostly looks stubbed-out on the NRF52 branch)

sslupsky, 2020-07-20:
@facchinm Why on earth is the boot loader source code not available on GitHub?  I've run into an issue with the MKR WAN 1310.  The boot loader the comes preinstalled with the hardware is causing excessive current consumption.  I am disappointed with this experience because it took much longer than it should have to isolate the problem.  Having access to the boot loader source would have saved considerable time.

I recommend you ship all Arduino boards with the SWD connector installed so we can easily replace the boot loader.

sabas1080, 2020-07-20:
We have port this core to be compatible with the bootloader UF2, if you want to test, is available here https://github.com/ElectronicCats/ArduinoCore-nRF528x-mbedos, UF2 is open source now

sslupsky, 2020-07-20:
Yes, I am using UF2.  That was how I discovered the problem.  I had developed my app on a board I was using with UF2.  Everything worked fine.  When I tried to assemble and flash a set of "new" boards, I used the Arduino boot loader (flashing UF2 on the MKR WAN requires adding another connector to the MKR WAN and using SWD).

I noticed the power consumption was 100x what it was supposed to be for the new set of boards.  I didn't realize at first that the problem was the boot loader ... I went searching for other sources of the problem, hardware differences between v2 and v3 boards, firmware issues with my app, etc.  I finally realized that the only difference was the boot loader.  So, I reflashed one of the new boards with UF2 and viola, works like a charm.

What concerns me is the time required to track the problem down to the boot loader. And,  I would like to avoid having to reflash the Arduino's with UF2.  Preferably, Arduino should be using UF2 by now and shipping their products with UF2.

Thanks for the link.  I'm am working towards migrating to nRF into my next products to avoid power consumption problems with Atmel SAMD/E.

rpavlik, 2020-07-21:
I have also used (the Adafruit) UF2 bootloader with this, which is why I wanted this source initially, to add compatibility with the BOSSA protocol the Arduino IDE uses by default - the alternative is having to create a separate or forked core.

I was very surprised when I got the Nano 33 BLE that Arduino hasn't moved to UF2 yet.

sslupsky, 2020-07-21:
Indeed.  I have also determined there are other issues with the Arduino boot loader that affect my application.  I have also observed there is something going on with respect to the RTC on SAMD boards that causes problems when waking from sleep.  Since I have no way to debug what is going because we have no source code I cannot explain what is going on.  What I can suggest is that experimentally, UF2 appears to eliminate this issue.

I was surprised by these observations as it is not obvious to me how the boot loader could be affecting the RTC and power consumption.  Nevertheless, it does appear to do so and UF2 appears to not have these undesirable side effects.

I tried the UF2 "update sketch" to update the boot loader but that bricked my board.  So, I had to attach wires so I could attach a SWD debugger and flash the new boot loader.  Have you had success updating the boat loader using the "update sketch" that the UF2 build process generates?

rpavlik, 2020-07-21:
No, unfortunately, I didn't even notice that it created an "update sketch" for this board. I built a little fixture with pogo pins to connect a debugger to the SWD pads to switch between the two bootloaders (and, because I didn't bother modifying the Arduino-supplied core, switching between BSP's)

per1234, 2020-07-22:
@sslupsky. This issue is about the Arduino Nano 33 BLE bootloader. Discussing problems with the bootloader of SAMD boards like your MKR WAN 1320 is off topic here.

As for the source code of that bootloader, it's right here:
https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero

bradjc, 2020-11-17:
Does anyone have a guide on how to replace the bootloader on the nano33 with a better one (without J-Link)?

bradjc, 2021-01-13:
Since the bossa bootloader is not open source, I developed a method for replacing the bootloader: https://github.com/tock/tock/blob/master/boards/nano33ble/README.md#getting-started

The bootloader we use for TockOS is custom, but mostly works like any other bootloader. Maybe it could be useful to someone even if not using Tock. In any case, it is open source: https://github.com/tock/tock-bootloader.