arduino/ArduinoCore-nRF528x-mbedos #24: Very slow compilation

fgaetani, 2019-10-12:
Hi I report a problem of slow compilation for this board Arduino NANO 33 BLE, especially if you use heavy libraries like ArduinoBLE. I noticed this problem after upgrading to MacOS Catalina from Mojave. Before the upgrade, the compilation lasted 10 seconds, now it lasts almost 1 minute. I tried to do the same test in Windows also using different versions of the IDE (1.8.10 and 1.9.0), getting worst results (5 minute for compiling). Is it possible to speed up compilation in any way? What has changed from MacOS Mojave to Catalina?

Thank you!

facchinm, 2019-10-12:
Hi @fgaetani, the slowdown on Catalina is frankly unexplainable (but that os is having so many issue this could be the least important).
About speeding up compilation, until mbed finishes the core structure rationalisation, we can just leverage Arduino caches to save time on subsequent compilations (the first one will be slow anyway).
Maybe take a shot of nightly IDE since it fixes a bug in library cache refusal, greatly speeding up library discovery phase.

fgaetani, 2019-10-12:
So would you advise me to try the nightly version of the IDE to try if the compilation speed improves?

facchinm, 2019-10-14:
Yes; the first compilation won't change at all but all subsequent ones should be much faster.
On windows the extreme slow speed could be due to an antivirus running in the background and uploading any temporary file before unlocking the file operation. Try pausing it (if you have one running of couser) and check if it improves the situation

fgaetani, 2019-10-14:
I have already tried to disable the antivirus on Windows not getting better results. However I work on MacOS, so the important thing is not to have problems with this OS. Will try the nightly version. Do I need to delete /User/username/Library /Arduino15 folder before installing the nightly version? Thank you

facchinm, 2019-10-14:
Not needed, you can just grab the hourly build from arduino.cc and run it .

fgaetani, 2019-10-14:
Thank you very much, the performance is much better with the nightly version.

closed 2019-10-14 by fgaetani