arduino/ArduinoCore-nRF528x-mbedos #25: Ports selection disabled

apas01, 2019-10-20:
Hi,
after succesfully working for some time, the "port" selection is now disabled (grey color).
I'm working on windows 10 (1803 version) and the Arduino IDE is 1.8.10.

The last thing performed was the upload of this sketch, that is probably not related.
`#include <Wire.h>
#include "SparkFun_BNO080_Arduino_Library.h"
BNO080 myIMU;
//====================================================IMU variables====================================================
struct Euler {
  float yaw;
  float pitch;
  float roll;
};
struct Quat {
  float i;
  float j;
  float k;
  float real;
};
Euler getAngles(Quat q, bool degrees);
Quat myQuat;
Euler eul;

//====================================================IMU variables====================================================

void setup()
{
  if (myIMU.begin() == false)
  {
    Serial.println("BNO080 not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...");
    while (1);
  }
  Wire.setClock(400000); //Increase I2C data rate to 400kHz

  myIMU.enableRotationVector(50); //Send data update every 50ms
}

void loop()
{
if (myIMU.dataAvailable() == true)
{
 float quatI = myIMU.getQuatI();
 float quatJ = myIMU.getQuatJ();
 float quatK = myIMU.getQuatK();
 float quatReal = myIMU.getQuatReal();
 float quatRadianAccuracy = myIMU.getQuatRadianAccuracy();

 myQuat.i = quatI;
 myQuat.j = quatJ;
 myQuat.k = quatK;
 myQuat.real = quatReal;
 eul = getAngles(myQuat);
 Serial.print(eul.yaw, 2);
 Serial.print(eul.pitch, 2);
 Serial.print(eul.roll, 2);
 Serial.print(quatRadianAccuracy, 2);
 Serial.println();
}
}
Euler getAngles(Quat q) {
Euler ret_val;
float x; float y;
x = 2 * ((q.i * q.j) + (q.real * q.k));
y = square(q.real) - square(q.k) - square(q.j) + square(q.i);
ret_val.yaw = degrees(atan2(y, x));
ret_val.pitch = degrees(asin(-2 * (q.i * q.k - q.j * q.real)));
x = 2 * ((q.j * q.k) + (q.i * q.real));
y = square(q.real) + square(q.k) - square(q.j) - square(q.i);
ret_val.roll = degrees(atan2(y , x));
return ret_val;
}
float square(float var) {
return var * var;
}`
In the device manager the family "PORT (COM and LPT)" is not present. I could restore legacy hardware but I don't see arduino nano 33 ble drivers in the arduino list.
Any idea?

facchinm, 2019-10-21:
Hi @apas01 ,
can you try double clicking the reset button? The led will start fading and the board will be in bootloader mode.
If the port appears in this case it means that some code you uploaded crashed it and there is probably some bug in the code

apas01, 2019-10-21:
You're correct, thank you.

closed 2019-10-21 by apas01