arduino/ArduinoCore-nRF528x-mbedos #26: Blink gets compile errors on Linux

dhalbert, 2019-10-26:
Using Arduino: 1.8.10 (tar file unpacked into personal `bin` directory)
Arduino nRF528x (Mbed OS) 1.1.2
on Ubuntu 18.04 x64

I get compilation errors when compiling the canonical Blink example. Blink compiles fine on Windows 10 with the same BSP and Arduino versions. After the first failure I removed `.arduino15` and reinstalled the BSP to make sure everything was clean, and the errors still occur.

```
Arduino: 1.8.10 (Linux), Board: "Arduino Nano 33 BLE"

/home/halbert/bin/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/halbert/bin/arduino-1.8.10/hardware -hardware /home/halbert/.arduino15/packages -tools /home/halbert/bin/arduino-1.8.10/tools-builder -tools /home/halbert/bin/arduino-1.8.10/hardware/tools/avr -tools /home/halbert/.arduino15/packages -built-in-libraries /home/halbert/bin/arduino-1.8.10/libraries -libraries /home/halbert/Arduino/libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path /tmp/arduino_build_565364 -warnings=none -build-cache /tmp/arduino_cache_113244 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.bossac.path=/home/halbert/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=/home/halbert/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.openocd.path=/home/halbert/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=/home/halbert/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -verbose /home/halbert/Arduino/Blink/Blink.ino
/home/halbert/bin/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/halbert/bin/arduino-1.8.10/hardware -hardware /home/halbert/.arduino15/packages -tools /home/halbert/bin/arduino-1.8.10/tools-builder -tools /home/halbert/bin/arduino-1.8.10/hardware/tools/avr -tools /home/halbert/.arduino15/packages -built-in-libraries /home/halbert/bin/arduino-1.8.10/libraries -libraries /home/halbert/Arduino/libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path /tmp/arduino_build_565364 -warnings=none -build-cache /tmp/arduino_cache_113244 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.bossac.path=/home/halbert/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=/home/halbert/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.openocd.path=/home/halbert/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=/home/halbert/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -verbose /home/halbert/Arduino/Blink/Blink.ino
Using board 'nano33ble' from platform in folder: /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2
Using core 'arduino' from platform in folder: /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2
Detecting libraries used...
/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -I. -w -g -Os -nostdlib @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/api/deprecated -iprefix/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_565364/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -I. -w -g -Os -nostdlib @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/api/deprecated -iprefix/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_565364/sketch/Blink.ino.cpp -o /tmp/arduino_build_565364/preproc/ctags_target_for_gcc_minus_e.cpp
/home/halbert/bin/arduino-1.8.10/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_565364/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -I. -w -g -Os -nostdlib @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE -I/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/api/deprecated -iprefix/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino @/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_565364/sketch/Blink.ino.cpp -o /tmp/arduino_build_565364/sketch/Blink.ino.cpp.o
In file included from /home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/functional:53:0,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/platform/cxxsupport/mstd_type_traits:1205,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/platform/mbed_atomic.h:884,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/features/netsocket/InternetSocket.h:28,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/features/netsocket/UDPSocket.h:23,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/features/netsocket/nsapi.h:40,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/mbed.h:27,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/Arduino.h:32,
                 from /tmp/arduino_build_565364/sketch/Blink.ino.cpp:1:
./new:1:40: error: redefinition of 'const uint8_t hid_report_descriptor [195]'
 const uint8_t hid_report_descriptor[195] = {
                                        ^
In file included from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/platform/Callback.h:22:0,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/rtos/Thread.h:29,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/rtos/rtos.h:28,
                 from /home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/mbed.h:23,
[... cut ...]
/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/rtos/Thread.h:253:68:   required from here
/home/halbert/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/mbed/platform/Callback.h:661:9: error: no matching function for call to 'operator new(sizetype, void*&)'
         new (d) F(*(F *)p);
         ^~~~~~~~~~~~~~~~~~
<built-in>: note: candidate: void* operator new(unsigned int)
<built-in>: note:   candidate expects 1 argument, 2 provided
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

facchinm, 2019-10-28:
Hi @dhalbert ,
are you sure there is no stray file in the folder where the Blink you are compiling lives?
The `./new:1:40: error:`  string looks suspicious.
I'll try on a clean 18.04 VM and will report back

dhalbert, 2019-10-28:
@facchinm Hmm, there is a file named `new`, but it is not in the `Blink` folder, it's in my home directory (!). It happens to contain some code with `hid_report_descriptor`. If I rename it to `new1`, the problem goes away. (There were files labeled `new` and `old` I was diff-ing.)

But this is now a different kind of bug. Why should a stray file named `new` be included in a compilation, and only for Nano 33 BLE? The panel icon I use to start arduino does start it up with my homedir as the current directory.

facchinm, 2019-10-28:
Indeed it's very suspicious; paths outside compiler and core folders should never be included... Unless you have something overriding the default paths as environment variable. Could it be the case?

dhalbert, 2019-10-28:
I don't see any such env var; perhaps there is some typo in the Mbed build scripts (like a `new` instead of `$new` or something).

dhalbert, 2019-10-28:
.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/functional:53 has these includes:

Note the `#include <new>`. So somehow `.` is part of the include path.

```
#if __cplusplus >= 201103L

#include <new>
#include <tuple>
#include <type_traits>
#include <bits/functional_hash.h>
#include <bits/invoke.h>
#include <bits/std_function.h>
#if __cplusplus > 201402L
# include <unordered_map>
# include <vector>
# include <array>
# include <utility>
# include <bits/stl_algo.h>
#endif
```

dhalbert, 2019-10-28:
For Arduino Nano 33 BLE< note, the `-I.`:
```
/home/halbert/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -I. ...
```
In a vanilla Uno build, there is no `-I.` as a command-line arg on any of the compilation lines.

dhalbert, 2019-10-28:
Sorry for the slew of messages. In `.arduino15/packages/arduino/hardware/mbed/1.1.2/platform.txt`, there is:
```
...
compiler.c.flags=-c -I. {compiler.warning_flags} -Os -g -nostdlib "@{compiler.mbed.defines}" "@{compiler.mbed.cflags}" {compiler.mbed.arch.define} -MMD -mcpu={build.mcu}
...
compiler.cpp.flags=-c -I. {compiler.warning_flags} -g -Os -nostdlib "@{compiler.mbed.defines}" "@{compiler.mbed.cxxflags}" {compiler.mbed.arch.define} -MMD -mcpu={build.mcu}
...
```

No other BSP `platform.txt` files that I have installed include `-I.`

facchinm, 2019-10-28:
You are TOTALLY right. I don't know when that include path slipped into (it exist since the initial commit).
I'm removing it right now; thank you so much for spotting this!

closed 2019-10-28 by facchinm