arduino/ArduinoCore-nRF528x-mbedos #27: Library Keypad i2c for Arduino Nano 33 (Mbed OS)

Ixso88, 2019-10-28:
I ask for help to insert the Keypad library i2c on the arduino nano 33, I tried but it gives me this error.I think there is a problem with enabling the Mbed os architecture to use wire.h pins.
```
In file included from C:\Users\Indagini Forensi 2\Documents\Arduino\libraries\Keypad_I2C\examples\CustomKeypad_I2C\CustomKeypad_I2C.ino:11:0:

C:\Users\Indagini Forensi 2\Documents\Arduino\libraries\Keypad_I2C/Keypad_I2C.h: In constructor 'Keypad_I2C::Keypad_I2C(char*, byte*, byte*, byte, byte, byte, byte)':

C:\Users\Indagini Forensi 2\Documents\Arduino\libraries\Keypad_I2C/Keypad_I2C.h:45:48: error: no matching function for call to 'arduino::MbedI2C::MbedI2C()'

   Keypad(userKeymap, row, col, numRows, numCols) { i2caddr = address; i2cwidth = width;}

                                                ^

In file included from C:\Users\Indagini Forensi 2\Documents\Arduino\libraries\Keypad_I2C/Keypad_I2C.h:36:0,

                 from C:\Users\Indagini Forensi 2\Documents\Arduino\libraries\Keypad_I2C\examples\CustomKeypad_I2C\CustomKeypad_I2C.ino:11:

C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:30:5: note: candidate: arduino::MbedI2C::MbedI2C(int, int)

     MbedI2C(int sda, int scl);

     ^~~~~~~

C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:30:5: note:   candidate expects 2 arguments, 0 provided

C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:27:7: note: candidate: constexpr arduino::MbedI2C::MbedI2C(const arduino::MbedI2C&)

 class MbedI2C : public HardwareI2C

       ^~~~~~~

C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:27:7: note:   candidate expects 1 argument, 0 provided

C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:27:7: note: candidate: constexpr arduino::MbedI2C::MbedI2C(arduino::MbedI2C&&)

C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:27:7: note:   candidate expects 1 argument, 0 provided

Più di una libreria trovata per "Keypad_I2C.h"
Usata: C:\Users\Indagini
Più di una libreria trovata per "Keypad.h"
Usata: C:\Users\Indagini
Più di una libreria trovata per "Wire.h"
Usata: C:\Users\Indagini
exit status 1
Errore durante la compilazione per la scheda Arduino Nano 33 BLE.
```

Ixso88, 2019-10-29:
Hi, I was referring to that library, because I didn't find other I2C Keypad libraries running on Mbed OS.However I have tried to modify the .cpp file twowire with wire but it continues to give me problems.

`Arduino:1.8.10 (Windows 10), Scheda:"Arduino Nano 33 BLE"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Indagini Forensi 2\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path C:\Users\INDAGI~1\AppData\Local\Temp\arduino_build_617942 -warnings=none -build-cache C:\Users\INDAGI~1\AppData\Local\Temp\arduino_cache_191878 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.bossac.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\INDAGI~1\AppData\Local\Temp\arduino_modified_sketch_9105\CustomKeypad_I2C.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Indagini Forensi 2\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path C:\Users\INDAGI~1\AppData\Local\Temp\arduino_build_617942 -warnings=none -build-cache C:\Users\INDAGI~1\AppData\Local\Temp\arduino_cache_191878 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.bossac.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\INDAGI~1\AppData\Local\Temp\arduino_modified_sketch_9105\CustomKeypad_I2C.ino
Using board 'nano33ble' from platform in folder: C:\Users\Indagini
Using core 'arduino' from platform in folder: C:\Users\Indagini
Detecting libraries used...
"C:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -I. -w -g -Os -nostdlib "@C:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\cores\\arduino" "-IC:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\cores\\arduino" "@C:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.2\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\INDAGI~1\\AppData\\Local\\Temp\\arduino_build_617942\\sketch\\CustomKeypad_I2C.ino.cpp" -o nul
Alternatives for Keypad_I2C.h: [Keypad_I2C]
ResolveLibrary(Keypad_I2C.h)
  -> candidates: [Keypad_I2C]
"C:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -I. -w -g -Os -nostdlib "@C:\\Users\\Indagini Forensi 2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware [... cut ...]
C:\Users\Indagini Forensi 2\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire/Wire.h:27:7: note:   candidate expects 1 argument, 0 provided

Più di una libreria trovata per "Keypad.h"
Usata: C:\Users\Indagini
Più di una libreria trovata per "Wire.h"
Usata: C:\Users\Indagini
Più di una libreria trovata per "Keypad_I2C.h"
Usata: C:\Users\Indagini
Uso la libreria Keypad_I2C nella cartella: C:\Users\Indagini Forensi
Uso la libreria Keypad nella cartella: C:\Users\Indagini Forensi
Uso la libreria Wire nella cartella: C:\Users\Indagini Forensi
exit status 1
Errore durante la compilazione per la scheda Arduino Nano 33 BLE.`

facchinm, 2019-10-29:
here's the fixed library 
[Keypad_I2C.zip](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/3783818/Keypad_I2C.zip)

Patch attached too :slightly_smiling_face: 
```patch
diff --git a/Keypad_I2C/Keypad_I2C.cpp b/Keypad_I2C/Keypad_I2C.cpp
index 2bd171e..7d7cf6d 100644
--- a/Keypad_I2C/Keypad_I2C.cpp
+++ b/Keypad_I2C/Keypad_I2C.cpp
@@ -34,14 +34,14 @@
 // Let the user define a keymap - assume the same row/column count as defined in constructor
 void Keypad_I2C::begin(char *userKeymap) {
     Keypad::begin(userKeymap);
-       TwoWire::begin();
+       Wire.begin();
        pinState = pinState_set( );
 }
 
 
 // Initialize I2C
 void Keypad_I2C::begin(void) {
-       TwoWire::begin();
+       Wire.begin();
 //     pinState = 0xff;
        pinState = pinState_set( );
 }
@@ -49,7 +49,7 @@ void Keypad_I2C::begin(void) {
 // Initialize I2C
 void Keypad_I2C::begin(byte address) {
        i2caddr = address;
-       TwoWire::begin(address);
+       Wire.begin();
 //     pinState = 0xff;
        pinState = pinState_set( );
 }
@@ -57,7 +57,7 @@ void Keypad_I2C::begin(byte address) {
 // Initialize I2C
 void Keypad_I2C::begin(int address) {
        i2caddr = address;
-       TwoWire::begin(address);
+       Wire.begin();
 //     pinState = 0xff;
        pinState = pinState_set( );
 }
@@ -76,10 +76,10 @@ void Keypad_I2C::pin_write(byte pinNum, boolean level) {
 
 int Keypad_I2C::pin_read(byte pinNum) {
        word mask = 0x1<<pinNum;
-       TwoWire::requestFrom((int)i2caddr, (int)i2cwidth);
-       word pinVal = TwoWire::read( );
+       Wire.requestFrom((int)i2caddr, (int)i2cwidth);
+       word pinVal = Wire.read( );
        if (i2cwidth > 1) {
-               pinVal |= TwoWire::read( ) << 8;
+               pinVal |= Wire.read( ) << 8;
        } 
        pinVal &= mask;
        if( pinVal == mask ) {
@@ -90,20 +90,20 @@ int Keypad_I2C::pin_read(byte pinNum) {
        word mask = 0x1<<pinNum;
-       TwoWire::requestFrom((int)i2caddr, (int)i2cwidth);
-       word pinVal = TwoWire::read( );
+       Wire.requestFrom((int)i2caddr, (int)i2cwidth);
+       word pinVal = Wire.read( );
        if (i2cwidth > 1) {
-               pinVal |= TwoWire::read( ) << 8;
+               pinVal |= Wire.read( ) << 8;
        } 
        pinVal &= mask;
        if( pinVal == mask ) {
@@ -90,20 +90,20 @@ int Keypad_I2C::pin_read(byte pinNum) {
 }
 
 void Keypad_I2C::port_write( word i2cportval ) {
-       TwoWire::beginTransmission((int)i2caddr);
-       TwoWire::write( i2cportval & 0x00FF);
+       Wire.beginTransmission((int)i2caddr);
+       Wire.write( i2cportval & 0x00FF);
        if (i2cwidth > 1) {
-               TwoWire::write( i2cportval >> 8 );
+               Wire.write( i2cportval >> 8 );
        }
-       TwoWire::endTransmission();
+       Wire.endTransmission();
        pinState = i2cportval;
 } // port_write( )
 
 word Keypad_I2C::pinState_set( ) {
-       TwoWire::requestFrom( (int)i2caddr, (int)i2cwidth );
-       pinState = TwoWire::read( );
+       Wire.requestFrom( (int)i2caddr, (int)i2cwidth );
+       pinState = Wire.read( );
        if (i2cwidth > 1) {
-               pinState |= TwoWire::read( ) << 8;
+               pinState |= Wire.read( ) << 8;
        }
        return pinState;
 } // set_pinState( )
diff --git a/Keypad_I2C/Keypad_I2C.h b/Keypad_I2C/Keypad_I2C.h
index e0af197..27b8eae 100644
--- a/Keypad_I2C/Keypad_I2C.h
+++ b/Keypad_I2C/Keypad_I2C.h
@@ -38,7 +38,7 @@
 #define        PCF8574 1       // PCF8574 I/O expander device is 1 byte wide
 #define PCF8575 2      // PCF8575 I/O expander device is 2 bytes wide
 
-class Keypad_I2C : public Keypad, public TwoWire {
+class Keypad_I2C : public Keypad {
 public:
        Keypad_I2C(char* userKeymap, byte* row, byte* col, byte numRows, byte numCols, byte address, byte width = 1) :
                Keypad(userKeymap, row, col, numRows, numCols) { i2caddr = address; i2cwidth = width;}  
```

closed 2019-10-29 by facchinm

Ixso88, 2019-10-30:
Thank you so much!