arduino/ArduinoCore-nRF528x-mbedos #29: sed: can't read cores/arduino/mbed/mbed.h: No such file or directory

matsujirushi, 2019-11-14:
I got error message below when run the mbed-os-to-arduino.

```
+ sed -i s#drivers/SPI.h#drivers/SPIMaster.h#g cores/arduino/mbed/mbed.h
sed: can't read cores/arduino/mbed/mbed.h: No such file or directory
```

It on `/tmp/mbed-os-program`.
Please check that.

facchinm, 2019-11-14:
Hi @matsujirushi ,
we are currently refactoring the script completely, please hold on a couple of days (or hours) and it will be new and shiny :slightly_smiling_face:

facchinm, 2019-11-14:
Just merged, the new script should work way better (and it's also compatible with macOS)

closed 2019-11-14 by facchinm

matsujirushi, 2019-11-15:
Thanks.

FYI
I can compile on Debian.
```
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/softdevice/common/nrf_sdh_ant.h
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/softdevice/common/nrf_sdh_ble.h
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/softdevice/common/nrf_sdh_soc.h
total: matches=0  hash_hits=0  false_alarms=0 data=5221231

sent 843,955 bytes  received 3,791 bytes  1,695,492.00 bytes/sec
total size is 5,221,231  speedup is 6.16
 done.
Generating flags... done.
Generating libs... done.
Copying generic MbedOS headers to core...  done.
Patching 'mbed.h'... done.
Patching SPI headers... done.

[mbed-os-to-arduino] MbedOS core generation ended succesfully.
```