arduino/ArduinoCore-nRF528x-mbedos #30: digitalRead forces INPUT_PULLUP

dcuartielles, 2019-11-14:
When creating a simple program to read a button with a pull-down resistor (10k) the input registers no voltage for as long as digitalRead is not called. Once digitalRead is issued, the pin seems to go into internal pull-up configuration, what makes the pin to be always up for the above-mentioned pull-down resistor.

**Reproduce the error**

1. Investigating the issue requires using a multimeter and making a simple program:

`void setup() {`
`  pinMode(2, INPUT);`
`}`
`    `
`void loop() {`
`}`

On the multimeter this reads: 0V

2. Change the program to contain a digitalRead():

`void setup() {`
`  pinMode(2, INPUT);`
`}`
`       `
`void loop() {`
`  digitalRead(2);`
`}`

The multimeter goes then into 3v3 for that pin.

On the other hand, when checking into the way digitalRead is defined, we find that it returns:

` return (PinStatus)mbed::DigitalIn(digitalPinToPinName(pin)).read();`

(as seen at: https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/cores/arduino/wiring_digital.cpp#L79)

This call to _DigitalIn_ seems to be lacking _PullNone_, what could be the reason for this not to work. However, this requires a deeper revision, as just adding PullNone, would force it also in the case when we would like _INPUT_PULLUP_ to be active.

facchinm, 2019-11-26:
@dcuartielles may you try this PR https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/31 and report any abnormal behaviour?

lorenzoromagnoli, 2020-01-29:
As reported [here](https://forum.arduino.cc/index.php?topic=661453.new;topicseen#new) I also experienced the same issue. 
I tested the PR, and indeed it fixes the wrong behaviour.

rmlearney-digicatapult, 2020-03-25:
I've tested this with PR #31 and it works properly (0V not 3.3V)

morettigiorgio, 2020-04-09:
It works for me too

closed 2020-07-30 by facchinm