arduino/ArduinoCore-nRF528x-mbedos #32: Is Wire library working as an I2C Slave?

kevingre, 2019-11-24:
I can't seem to get a Nano 33 BLE configured as a slave.  I see in the Wire.cpp that DEVICE_I2CSLAVE needs to be defined, but when I do that, it will not compile.  Just checking to see Wire under mbedos SHOULD be working as a slave.

facchinm, 2019-11-29:
Hi @kevingre ,
AFAIK there's no plan yet to add I2C slave support to mbed's nrf52 (should go here https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/i2c_api.c#L935 ) but I'll investigate and get back to you

brendansmallco, 2020-07-14:
Any word on this? Seeing a lot of problems across Nano 33 and ESP32 without this functionality.

facchinm, 2020-07-14:
mbed core 1.2.x (you can find it searching for `Portenta` on Board Manager) contains the Wire slave implementation for Nano33BLE too. It is not yet official due to some much needed testing but you can test it immediately.
Any report is appreciated :slightly_smiling_face: