arduino/ArduinoCore-nRF528x-mbedos #36: Raspberry PI: unrecognized command line option '-std=gnu++14'

gdsports, 2019-12-04:
Raspberry Pi 4 running Raspbian buster.

`Linux rpi44gb 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`

```
Arduino: 1.8.10 (Linux), Board: "Arduino Nano 33 BLE"

/home/pi/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/pi/arduino-1.8.10/hardware -hardware /home/pi/.arduino15/packages -tools /home/pi/arduino-1.8.10/tools-builder -tools /home/pi/arduino-1.8.10/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/arduino-1.8.10/libraries -libraries /home/pi/Arduino/libraries -fqbn=arduino:mbed:nano33ble -vid-pid=0000_0000 -ide-version=10810 -build-path /tmp/arduino_build_113679 -warnings=none -build-cache /tmp/arduino_cache_994791 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -verbose /home/pi/Arduino/blink/blink.ino
/home/pi/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/pi/arduino-1.8.10/hardware -hardware /home/pi/.arduino15/packages -tools /home/pi/arduino-1.8.10/tools-builder -tools /home/pi/arduino-1.8.10/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/arduino-1.8.10/libraries -libraries /home/pi/Arduino/libraries -fqbn=arduino:mbed:nano33ble -vid-pid=0000_0000 -ide-version=10810 -build-path /tmp/arduino_build_113679 -warnings=none -build-cache /tmp/arduino_cache_994791 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -verbose /home/pi/Arduino/blink/blink.ino
Using board 'nano33ble' from platform in folder: /home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3
Using core 'arduino' from platform in folder: /home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3
Detecting libraries used...
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/defines.txt @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated -iprefix/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_113679/sketch/blink.ino.cpp -o /dev/null
Error while detecting libraries included by /tmp/arduino_build_113679/sketch/blink.ino.cpp
Generating function prototypes...
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/defines.txt @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated -iprefix/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_113679/sketch/blink.ino.cpp -o /tmp/arduino_build_113679/preproc/ctags_target_for_gcc_minus_e.cpp
arm-none-eabi-g++: error: unrecognized command line option '-std=gnu++14'
exit status 1
Error compiling for board Arduino Nano 33 BLE.

```

facchinm, 2019-12-04:
Hi @gdsports ,
we've been testing a proper gcc7 build for arm32 platforms for quite some time and it's working fine. I just pushed it in production so in a couple of hours you'll just need to uninstall and reinstall Nano 33 BLE core and it will automatically download the new toolchain.

gdsports, 2019-12-04:
Thanks, the updated toolchain fixes the "-std=gcc++14" error. I removed ~/.arduino15 to force the IDE to download the latest packages and archives. But the build fails with "libfl.so.2: cannot open shared object file: No such file or directory".

```
Arduino: 1.8.10 (Linux), Board: "Arduino Nano 33 BLE"

/home/pi/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/pi/arduino-1.8.10/hardware -hardware /home/pi/.arduino15/packages -tools /home/pi/arduino-1.8.10/tools-builder -tools /home/pi/arduino-1.8.10/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/arduino-1.8.10/libraries -libraries /home/pi/Arduino/libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path /tmp/arduino_build_63265 -warnings=none -build-cache /tmp/arduino_cache_345336 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.bossac.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -verbose /home/pi/arduino-1.8.10/examples/01.Basics/Blink/Blink.ino
/home/pi/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/pi/arduino-1.8.10/hardware -hardware /home/pi/.arduino15/packages -tools /home/pi/arduino-1.8.10/tools-builder -tools /home/pi/arduino-1.8.10/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/arduino-1.8.10/libraries -libraries /home/pi/Arduino/libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path /tmp/arduino_build_63265 -warnings=none -build-cache /tmp/arduino_cache_345336 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs=runtime.tools.bossac.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -verbose /home/pi/arduino-1.8.10/examples/01.Basics/Blink/Blink.ino
Using board 'nano33ble' from platform in folder: /home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3
Using core 'arduino' from platform in folder: /home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3
Detecting libraries used...
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/defines.txt @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated -iprefix/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_63265/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/defines.txt @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE -I/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino/api/deprecated -iprefix/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/cores/arduino @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_63265/sketch/Blink.ino.cpp -o /tmp/arduino_build_63265/preproc/ctags_target_for_gcc_minus_e.cpp
/home/pi/arduino-1.8.10/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_63265/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pi/.arduino15/packages/arduino/hardware/mbed/1.1.3/variants/ARDUINO_NANO33BLE/defines.txt @/home/pi/.arduino15/packages/ard [... cut ...] no_build_63265/core/mbed/platform/cxxsupport/mstd_mutex.cpp.o
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar rcs /tmp/arduino_build_63265/core/core.a /tmp/arduino_build_63265/core/itoa.c.o
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: error while loading shared libraries: libfl.so.2: cannot open shared object file: No such file or directory
exit status 127
Error compiling for board Arduino Nano 33 BLE.

```

gdsports, 2019-12-04:
I installed libfl2 to get the missing shared library libfl.so.2 but this might not be the correct solution since it requires root privs.

`sudo apt install libfl2`

The IDE finished the compile and upload without errors. 

Perhaps the solution should be to bundle libfl.so.2 in the arm gcc lib directory.
```
pi@rpi4rm201a:~/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/lib $ ls -l
total 640
drwxr-xr-x 3 pi pi   4096 Nov 11 02:13 gcc
-rwxr-xr-x 1 pi pi    967 Nov 11 05:53 libcc1.la
lrwxrwxrwx 1 pi pi     15 Dec  4 09:21 libcc1.so -> libcc1.so.0.0.0
lrwxrwxrwx 1 pi pi     15 Dec  4 09:21 libcc1.so.0 -> libcc1.so.0.0.0
-rwxr-xr-x 1 pi pi 645656 Nov 11 05:53 libcc1.so.0.0.0
```

facchinm, 2019-12-05:
I agree that all the dependencies should be self contained; however I used the very same build script ARM uses to generate its official toolchain, so an additional dependency is, well, unexpected.
Running `ldd` on my Linux x86_64 installation gives 
```
ldd ~/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar
	linux-vdso.so.1 (0x00007ffd5f772000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f70aeb0a000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f70ae943000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f70aeb6d000)
```
So no sign of `libfl2` dependency. If you run the same command are you getting different results?

gdsports, 2019-12-05:
Yes, libfl and libm. Perhaps the ARM version is built with different options.

```
pi@rpi44gb:~ $ ldd ~/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar
	linux-vdso.so.1 (0xbeeec000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6f22000)
	libfl.so.2 => /usr/lib/arm-linux-gnueabihf/libfl.so.2 (0xb6ef6000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6ee3000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6d95000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6d13000)
	/lib/ld-linux-armhf.so.3 (0xb6f37000)

```

facchinm, 2019-12-05:
`objdump --dynamic-syms arm-none-eabi-ar` shows that the dependency is due to `yylex` function. Not 100% sure but I think the Raspbian installation I used to build the toolchain didn't include a static library for `libfl`, so the dynamic one has been picked up.

gdsports, 2019-12-05:
Static linking libs makes a lot of sense. 

I noticed one oddity that may not be relevant. Looks like the --version option reports 7-2019-q4-major. Maybe a typo?

```
pi@rpi44gb:~ $ ~/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar --version
GNU ar (GNU Tools for Arm Embedded Processors 7-2019-q4-major) 2.29.51.20171128
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.

```

facchinm, 2019-12-05:
On our own build scripts we always static link everything to avoid exactly this kind of issues. However, this build script comes directly from arm (https://developer.arm.com/-/media/Files/downloads/gnu-rm/7-2017q4/gcc-arm-none-eabi-7-2017-q4-major-src.tar.bz2?revision=407b9d39-904c-4294-af9c-3b0cdd04184b?product=GNU%20Arm%20Embedded%20Toolchain,Invariant,,Source,7-2017-q4-major) and has the peculiar "feature" to change the "date" field based on when you compile it (thus `2019-q4`)

Hutkikz, 2019-12-15:
Just a note that I too had this problem except that I am using a MKR WiFi 1010.
@gdsports solution also worked for me .