arduino/ArduinoCore-nRF528x-mbedos #37: Wire library, i2C NOT WORKING

fgaetani, 2019-12-04:
There are many problems with the Wire library. I tried to run different sketches on two Arduino Nano BLE 33 boards (one master and one slave), including the examples, but the communication does not work. Specifically, the onRequest() and onReceive() functions do not seem to work. Furthermore, it is not possible to send a string with the Wire.write() function, as could be done with the AVR platform. I analyzed the signals with a logic analyzer and for example in the case of the requestFrom() I do not receive any response from the slave. How can the problem be solved?

fgaetani, 2019-12-04:
I solved using an Arduino MKR 1010 as a slave and an Arduino Nano BLE 33 as master.The problem seems to be in the Wire.begin(slave_address) function in slave sketch. The devices with salve_address does not send ACK, then the connection goes down. I hope for a solution. Thanks

facchinm, 2019-12-04:
Hi @fgaetani ,
the Wire slave implementation is incomplete at the moment, an the callbacks are empty.
This is due to https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/32 .
Let's keep track of the issue there, thanks.

closed 2019-12-04 by facchinm