arduino/ArduinoCore-nRF528x-mbedos #38: [Feature Request] Add SoftwareSerial / Uart lib to Nano 33

pierre-b, 2019-12-16:
Hello Arduino,

For an IoT project I need to connect a GPS and GPRS board, each needs a Serial port.

As there is only one Serial port available (the USB is taken) I need a second Serial port. Unfortunately SoftwareSerial and Uart libs don't work on Nano 33 and I'm stuck.

Is it something you could add? or clearly I can't build my project with a Nano 33?

Thanks

ping @ostaquet

pierre-b, 2019-12-16:
Hi @facchinm , yes I meant Serial1 when I said:

> there is only one Serial port available

So I would need a Serial2 or SoftwareSerial / Uart libs. I didn't express myself clearly.

Are SoftwareSerial / Uart libs a possibility in the near future or should I ditch the Nano 33 BLE?

Thanks

facchinm, 2019-12-16:
Got it, sorry for the confusion. nRF52480 has support for two hardware UART; we are using just one (on pins 0 and 1) but you can easily instantiate another.
Two methods: changing the core with the following patch
```patch
diff --git a/variants/ARDUINO_NANO33BLE/defines.txt b/variants/ARDUINO_NANO33BLE/defines.txt
index e995d15e..2365b15e 100644
--- a/variants/ARDUINO_NANO33BLE/defines.txt
+++ b/variants/ARDUINO_NANO33BLE/defines.txt
@@ -21,7 +21,7 @@
 -DDEVICE_PORTINOUT=1
 -DDEVICE_PORTOUT=1
 -DDEVICE_PWMOUT=1
--DDEVICE_SERIAL=1
+-DDEVICE_SERIAL=2
 -DDEVICE_SERIAL_ASYNCH=1
 -DDEVICE_SERIAL_FC=1
 -DDEVICE_SLEEP=1
diff --git a/variants/ARDUINO_NANO33BLE/pins_arduino.h b/variants/ARDUINO_NANO33BLE/pins_arduino.h
index 5937dba1..3ff156e6 100644
--- a/variants/ARDUINO_NANO33BLE/pins_arduino.h
+++ b/variants/ARDUINO_NANO33BLE/pins_arduino.h
@@ -60,6 +60,9 @@ static const uint8_t A7  = PIN_A7;
 #define PIN_SERIAL_RX (1ul)
 #define PIN_SERIAL_TX (0ul)
 
+#define PIN_SERIAL1_RX (3ul)
+#define PIN_SERIAL1_TX (4ul)
+
 // SPI
 #define PIN_SPI_MISO  (12u)
 #define PIN_SPI_MOSI  (11u)
@@ -111,9 +114,11 @@ static const uint8_t SCK  = PIN_SPI_SCK;
 
 
 // Mbed specific defines
-#define SERIAL_HOWMANY         1
+#define SERIAL_HOWMANY         2
 #define SERIAL1_TX                     (digitalPinToPinName(PIN_SERIAL_TX))
 #define SERIAL1_RX                     (digitalPinToPinName(PIN_SERIAL_RX))
+#define SERIAL2_TX                     (digitalPinToPinName(PIN_SERIAL1_TX))
+#define SERIAL2_RX                     (digitalPinToPinName(PIN_SERIAL1_RX))
 
 #define SERIAL_CDC                     1
 #define HAS_UNIQUE_ISERIAL_DESCRIPTOR
```
(you can change PIN_SERIAL1_RX / PIN_SERIAL1_TX as you prefer :slightly_smiling_face: )

or from the code, inside the sketch
```
UART mySerial(digitalPinToPinName(4), digitalPinToPinName(3), NC, NC);
```
(change 4 and 3 as you prefer, the processor can mux any functionality on any pin)

pierre-b, 2019-12-17:
Thank you @facchinm , will those modifications to the core be released officially?

Also, does the "UART mySerial(..." config work without modifiying the core?

facchinm, 2019-12-17:
The `UART mySerial(...` works without any core modification.
We will never release the core patch because it would limit user freedom to move the pins wherever s/he needs

closed 2019-12-17 by facchinm

zungun, 2020-09-28:
diff --git a/variants/ARDUINO_NANO33BLE/defines.txt b/variants/ARDUINO_NANO33BLE/defines.txt

where can i find this route??

Maximumotion, 2020-12-24:
Hello Arduino, 
I wanted to ask if the method of adding the second serial to Arduino Nano 33 BLE IoT is the same as describe in the above topic. 
I am connecting a Nextion Display to the board and also using GPS so trying to figure out how to add the second serial. 
thanks

facchinm, 2020-12-24:
Hi @Maximumotion ,
do you mean `Arduino Nano 33 IoT`? In that case, the reference method to add a second serial port is explained [here](https://www.arduino.cc/en/Tutorial/SamdSercom#create-a-new-serial-instance)

Maximumotion, 2020-12-25:
Thank you for your reply!
Yes, I meant Arduino Nano 33 IoT.
Just to make sure I understood it right as its quite technical. On the page with the explanation there was a reference to a variant.cpp file that is specific tot the board. 
Looking to the variant.cpp for Arduino Nano 33 IoT, I can see there are 2 serials: 
Uart Serial1( &sercom5, PIN_SERIAL1_RX, PIN_SERIAL1_TX, PAD_SERIAL1_RX, PAD_SERIAL1_TX ) ;
and 
Uart Serial2(&sercom3, PIN_SERIAL2_RX, PIN_SERIAL2_TX, PAD_SERIAL2_RX, PAD_SERIAL2_TX);

So by default, Arduino Nano 33 IoT has  Serial1 set to pin 0 and 1. 
So based on explained, If I want second Serial on pins 3 and 2, it would look something like this?

Uart mySerial (&sercom3, 3, 2, SERCOM_RX_PAD_3, UART_TX_PAD_2); 

Thank you!

facchinm, 2020-12-28:
@Maximumotion Nano33IoT is already using sercom 2,3,5 internally (https://github.com/arduino/ArduinoCore-samd/blob/master/variants/nano_33_iot/variant.cpp#L188-L207) , so you need to check [these](https://github.com/arduino/ArduinoCore-samd/blob/0b60a79c4b194ed2e76fead95caf1bbce8960049/variants/nano_33_iot/variant.cpp#L37-L38) lines to find the supported sercom (if it exists).
In this case we are lucky and it's sercom4, but you would lose I2C since it uses the same port (https://github.com/arduino/ArduinoCore-samd/blob/master/variants/nano_33_iot/variant.h#L154).

In the end, your extra serial port becomes
```
Uart mySerial (&sercom4, 3, 2, SERCOM_RX_PAD_3, UART_TX_PAD_2);
```

Maximumotion, 2020-12-28:
Thank you for your help!

Infinityme11, 2021-01-05:
Hi, I am using the Arduino Nano BLE 33 Sense,  and I want to use ### UART mySerial(digitalPinToPinName(4), digitalPinToPinName(3), NC, NC);. 

May I know the procedures of using the code as I do not understand the instruction above.

 Am I supposed to download a library? 

Thanks in advance