arduino/ArduinoCore-nRF528x-mbedos #39: Microcontroller freezes

fgaetani, 2019-12-16:
I have a problem with the Arduino Nano 33 BLE board with ArduinoBLE library. As soon as I try to connect to the device through the smartphone, the microcontroller freezes and the yellow LED flashes 5 times fast and 5 times slow and I am forced to reflash the firmware by double clicking the reset button. I have found similar problems also with the use of Wire library. What can this problem be due to?

fgaetani, 2019-12-18:
I solved using PlatformIO, the same project is compiled without any problems, it works on the microcontroller and no runtime errors occur. How is this possible? Does PlatformIO use the same compilers of Arduino IDE or am I wrong? It is paradoxical that official support works worse than unofficial support.

facchinm, 2020-03-10:
Hi @fgaetani ,
AFAIK platform.io uses our source code and compilers so I truly doubt the issue to be there. Maybe they are sticking to an older version of the core that doesn't exhibit the bug you are encountering?

fgaetani, 2020-03-10:
Hi @facchinm ,
it's been a long time, but I remember very well that the core version and the main code compiled were the same.

oacs, 2020-03-19:
When the yellow LED flashes 5 times fast and 5 times slow is a runtime error ?

fgaetani, 2020-03-23:
Hi @oacs, yes, that is the reporting of a runtime error.