arduino/ArduinoCore-nRF528x-mbedos #40: More than 3 concurrent BLE Connections (Arduino Nano 33 BLE)

Dorsel89, 2019-12-18:
Hi everyone,
I would like to have more than 3 concurrent BLE connections with the Arduino Nano 33 BLE. As far as i understand i have to adjust the DM_CONN_MAX in [cfg_stack.h](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/beac74ca3cd9d07363f66cf9cda6b143e4385cd2/cores/arduino/mbed/features/FEATURE_BLE/targets/TARGET_CORDIO/stack/ble-host/sources/stack/cfg/cfg_stack.h).

Unfortunately, that doesn't work, do you have any tips on what I'm doing wrong?
Thanks and kind regards!

sandeepmistry, 2020-01-02:
Hi @Dorsel89,

I believe the link layer inside [libmbed.a](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/variants/ARDUINO_NANO33BLE/libs/libmbed.a) will need to be re-compiled for the changes to take effect. I also believe [variants/ARDUINO_NANO33BLE/mbed_config.h](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/a1816cd20de39ffc03a9cfd6b378e7b63b210a89/variants/ARDUINO_NANO33BLE/mbed_config.h#L43) is the correct place to update the define.

There is a script to re-compile mbed: https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/mbed-os-to-arduino

Dorsel89, 2020-01-07:
Hi @sandeepmistry 
thanks for your help! I followed your suggestions but I am still not able to connect more devices (or less after I decreased _DM_CONN_MAX_).

I wonder about the mbed_config.h , in the header it is mentioned that the file should not be edited, because the values are generated automatically. This is exactly what I noticed, after I start the script _mbed-os-to-arduino_ all defines are back to default.

It would be nice if you have more tips for me!

sandeepmistry, 2020-01-09:
Hi @Dorsel89,

I've just spoken to @facchinm about this, the correct place to add the define is in [variants/ARDUINO_NANO33BLE/conf/mbed_app.json](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/variants/ARDUINO_NANO33BLE/conf/mbed_app.json#L2-L7), then re-compile mbed.

Apologies for the incorrect information I provided earlier.

Dorsel89, 2020-01-10:
Hey @sandeepmistry,

i think i still need some help :-/ I dont know how do edit the json file properly. I added something like:
`"cordio.dm_conn_max": 20
`
to the _target_overrides_ . But im not able to recompile afterwards.

Thanks in advance

Dorsel89, 2020-01-14:
Hi @sandeepmistry,
I would like to sum up my current results. I edited my [mbed_app.json](https://github.com/Dorsel89/ArduinoCore-nRF528x-mbedos/blob/7a08666e23f83287035a1d8c1d46473a916ebcfe/variants/ARDUINO_NANO33BLE/conf/mbed_app.json), recompiled everything, and be able to connect up to 8 devices to my arduino.

I found the right macro names in this file (mbed-os repository) : [mbed_lib.json](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/targets/TARGET_CORDIO/mbed_lib.json).

Unfortunately I cannot increase the maximum number of connections. If I set max-connections to something >8 the BLE Server does not start. Maybe I have to increse some memory size? If yes, where?  But I have to admit, I actually dont know how to debug this right now.

facchinm, 2020-01-14:
Hi @Dorsel89 ,
sorry for the late response but I missed this thread. It could indeed be a memory limitation related to ble thread; if so, you could try increasing the allocated memory here (https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/HCICordioTransport.cpp#L164) so that line reads
```
static rtos::Thread bleLoopThread(osPriorityNormal, PREFERRED_STACK_SIZE);
```
By increasing `PREFERRED_STACK_SIZE` (from the initial 4KB) you can check if the BLE server starts again.

Dorsel89, 2020-01-17:
Hey @facchinm,

I just checked if the Bluetooth Server starts by changeing:

A) increase the STACK_SIZE of the bleLoopThread
and 
B) increase the MBED_CONF_RTOS_THREAD_STACK_SIZE with the mbed_app.json

Of course both should do the same in the end, both ways do not work for max-connections = 9. I changed the size to 2*4096 for both ways.

Do you have any further ideas?

Many Thanks

gnsmrky, 2020-08-13:
@Dorsel89 Can you share how you managed to do multiple concurrent connections?  I have Arduino Nano 33 BLE acting as central.  Only got 1 peripheral connection. 2nd connection always fails.

Polldo, 2020-08-17:
@gnsmrky you can follow this issue https://github.com/arduino-libraries/ArduinoBLE/issues/108

Dorsel89, 2020-08-17:
Hey @gnsmrky  & @Polldo ,
sorry for my late response. You can take a look at this library:

https://github.com/Dorsel89/phyphox-arduino

I just tested the example "randomNumbers" and it works on 2 phones at the same time for me. It should work up to 3 concurrent connections without changing the mbed_app.json. If you change the mbed_app.json as discussed in this issue you can get up to 8 connections. Unfortunately i did not manage to get more connections but it is still on my todo list.

Hope this helps

Dorsel89, 2020-10-29:
Hey @giulcioffi,

is there reason to stick at 5 connections? In general I would like to get up to 20 connections but I couldnt figure out why my bluetoothserver crashed with DM_CONN_MAX>8. Unfortunately this topic is not on top of my TODO list so I didnt spent any time on this since months..

giulcioffi, 2020-10-30:
Hi @Dorsel89,
the reason why it crashes with DM_CONN_MAX = 8 is because there is not enough space in the RAM for it. We decided to stick at 5 connections because this can be handled fine with the RAM. However, since the methodology to increase the number of connected BLE peripherals has been provided, nothing prevents the user from customly adjust this number according to his/her needs :) Keep in mind that the higher the number of connected peripherals, the higher the [min/max time intervals in ArduinoBLE library](https://github.com/arduino-libraries/ArduinoBLE/pull/105/commits/8c6d1452362897d14c31d78b4dfb12a5aa1435a5). So if you decide to increase DM_CONN_MAX, remember to increase also these two connection parameters. The explanation for this and how to set these parameters comes from [mbed-os](https://github.com/arduino/mbed-os/blob/9fefb7db4fdfed229a3a053d3b53a99cac6664ef/features/FEATURE_BLE/targets/TARGET_CORDIO_LL/stack/controller/sources/ble/sch/sch_rm.c#L84).