arduino/ArduinoCore-nRF528x-mbedos #42: Use LoRaWAN / SX127x lora radio modules on the Nano BLE 33 (Sense)

gonzalocasas, 2020-01-03:
NOTE: This is a question rather than an issue. I [posted it on the forum](https://forum.arduino.cc/index.php?topic=634356.0), but got no replies, so I'm re-posting here! ;)

---

I am playing with the new Nano 33 BLE boards and I was very surprised when I read about these boards being built on top of mBed OS!

Now I would like to connect them to an SX127x radio module (like the RFM95) to be able to send/receive stuff over LoRaWAN (TTN), but I'm not entirely sure where to start on this.

It looks like the [loramac feature is part of the core](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/tree/master/cores/arduino/mbed/features/lorawan), but I am not really sure where and how to load the SX127x radio library (or if it's already part of mbedOS somehow). 

If there's any examples on how to get this to work, it would be greaaaaatly appreciated!!

facchinm, 2020-01-09:
Hi @gonzalocasas ,
you surely can :smile: It's just not supereasy. Let me sum it up.
* create a sketch with this content
```
#include "lorawan/LoRaWANInterface.h"
#include "lorawan/system/lorawan_data_structures.h"
#include "events/EventQueue.h"

// Application helpers
#include "SX1276_LoRaRadio.h"
#include "lorawan/lorastack/phy/LoRaPHYEU868.h"
#include "lorawan/lorastack/phy/LoRaPHYAS923.h"
#include "lorawan/lorastack/phy/LoRaPHYAU915.h"
#include "lorawan/lorastack/phy/LoRaPHYCN470.h"
#include "lorawan/lorastack/phy/LoRaPHYCN779.h"
#include "lorawan/lorastack/phy/LoRaPHYEU433.h"
#include "lorawan/lorastack/phy/LoRaPHYIN865.h"
#include "lorawan/lorastack/phy/LoRaPHYKR920.h"
#include "lorawan/lorastack/phy/LoRaPHYUS915.h"

SX1276_LoRaRadio radio(MBED_CONF_APP_LORA_SPI_MOSI,
                       MBED_CONF_APP_LORA_SPI_MISO,
                       MBED_CONF_APP_LORA_SPI_SCLK,
                       MBED_CONF_APP_LORA_CS,
                       MBED_CONF_APP_LORA_RESET,
                       MBED_CONF_APP_LORA_DIO0,
                       MBED_CONF_APP_LORA_DIO1,
                       MBED_CONF_APP_LORA_DIO2,
                       MBED_CONF_APP_LORA_DIO3,
                       MBED_CONF_APP_LORA_DIO4,
                       MBED_CONF_APP_LORA_DIO5,
                       MBED_CONF_APP_LORA_RF_SWITCH_CTL1,
                       MBED_CONF_APP_LORA_RF_SWITCH_CTL2,
                       MBED_CONF_APP_LORA_TXCTL,
                       MBED_CONF_APP_LORA_RXCTL,
                       MBED_CONF_APP_LORA_ANT_SWITCH,
                       MBED_CONF_APP_LORA_PWR_AMP_CTL,
                       MBED_CONF_APP_LORA_TCXO);

static LoRaPHYEU868 phy;
static LoRaWANInterface lorawan(radio, phy);

void setup() {
  // put your setup code here, to run once:

}
```
(you'll need to define the pins in mbed format, like `P0_3` , check the pinout diagram on https://store.arduino.cc/arduino-nano-33-ble and fill the unconnected pins with `NC`)

* Download the radio driver from https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers and slightly modify it (I'm attaching the modified version, ready to be copied to sketchbook/libraries [SX1276.zip](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/4042666/SX1276.zip)
)

* Enjoy :slightly_smiling_face: 

Let me know if it works in your setup, and if it turns out to be useful feel free to post it somewhere!

gonzalocasas, 2020-01-27:
@facchinm I will give it a try very soon! Thanks for the detailed write up!

facchinm, 2020-03-11:
Closing to keep the issue tracker tidy but feel free to post a writeup (here or anywhere else) when you are done :wink:

closed 2020-03-11 by facchinm

sabas1080, 2020-12-14:
Hi @facchinm 

I am picking up this issue, I am trying to use sx126x with this core

when using `NC` I have the next error:
```
lowaran_mbed:54:26: error: invalid conversion from 'unsigned int' to 'PinName' [-fpermissive]
                        NC);
```

facchinm, 2020-12-14:
Hi @sabas1080 ,
which core are you using? The extracted library should be updated for core 1.3.0 since some definitions changed in the meantime (so the zip attachment is no longer "valid"). Let me know if I can help in this.

sabas1080, 2020-12-14:
@facchinm I am using https://github.com/arduino/ArduinoCore-mbed

facchinm, 2020-12-14:
Here's the patched library.
[SX1276_1.zip](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/5687608/SX1276_1.zip)
If ANT_SWITCH is NC you will need to modify `MBED_CONF_SX1276_LORA_DRIVER_RADIO_VARIANT` macro in the top lines of `SX1276_LoRaRadio.cpp`
Let me know if it works

sabas1080, 2020-12-14:
Yeah! SX1276 compile, I have the SX126x, I have made similar changes, but I have the next error:

```
Arduino/libraries/SX126X/SX126X_LoRaRadio.cpp: In member function 'virtual void SX126X_LoRaRadio::standby()':
/Users/sabas/Documents/Arduino/libraries/SX126X/SX126X_LoRaRadio.cpp:373:28: error: 'MBED_CONF_SX126X_LORA_DRIVER_STANDBY_MODE' was not declared in this scope
     uint8_t standby_mode = MBED_CONF_SX126X_LORA_DRIVER_STANDBY_MODE;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126X/SX126X_LoRaRadio.cpp:373:28: note: suggested alternative: 'MBED_CONF_SX126X_LORA_DRIVER_RADIO_VARIANT'
     uint8_t standby_mode = MBED_CONF_SX126X_LORA_DRIVER_STANDBY_MODE;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                            MBED_CONF_SX126X_LORA_DRIVER_RADIO_VARIANT
/Users/sabas/Documents/Arduino/libraries/SX126X/SX126X_LoRaRadio.cpp: In member function 'void SX126X_LoRaRadio::cold_start_wakeup()':
/Users/sabas/Documents/Arduino/libraries/SX126X/SX126X_LoRaRadio.cpp:431:30: error: 'MBED_CONF_SX126X_LORA_DRIVER_REGULATOR_MODE' was not declared in this scope
     uint8_t regulator_mode = MBED_CONF_SX126X_LORA_DRIVER_REGULATOR_MODE;
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/sabas/Documents/Arduino/libraries/SX126X/SX126X_LoRaRadio.cpp:431:30: note: suggested alternative: 'MBED_CONF_SX126X_LORA_DRIVER_RADIO_VARIANT'
     uint8_t regulator_mode = MBED_CONF_SX126X_LORA_DRIVER_REGULATOR_MODE;
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              MBED_CONF_SX126X_LORA_DRIVER_RADIO_VARIANT
```

sabas1080, 2020-12-14:
Ok, ok i have seen this is declared in mbed_config.h, is ok?

https://github.com/ArrowElectronicsESC/SiLabsLoRa_EFR_BG/blob/master/mbed_config.h

complie for me, Should I add the other definitions?

facchinm, 2020-12-15:
You can add your personalized pins definitions at core level (so in mbed_config.h) or directly in the library, which is a bit cleaner and doesn't get overwritten when updating the core. Of course the best strategy would be patching the library to accept that values from the constructor, so you only have to modify the sketch to match your wiring :slightly_smiling_face:

sabas1080, 2020-12-20:
Ok, I have started an example where I have the changes that you have sent me

https://github.com/sabas1080/mbed-semtech-lora-rf-drivers

Based in:
https://github.com/ARMmbed/mbed-connect-2018-lorawan/blob/master/main.cpp 
and
https://github.com/maxgerhardt/pio-lorawan-example/blob/master/src/main.cpp

device does not start @facchinm

facchinm, 2020-12-21:
@sabas1080 which board are you using? If you can't communicate with the semtech module you can try using a lower-level library like https://github.com/sandeepmistry/arduino-LoRa to make sure that the wiring is correct.

sabas1080, 2020-12-21:
@facchinm  I'm using SX1268, I can communicate with the semtech module a lower-level with library https://github.com/StuartsProjects/SX12XX-LoRa

But with my example my board be break not start  and USBSerial or monitor disappears

facchinm, 2020-12-22:
@sabas1080 which board / core version are you on? Is there any "morse code of death" by the builtin LED?