arduino/ArduinoCore-nRF528x-mbedos #43: Library BQ25896 not working with Arduino Nano 33

morettigiorgio, 2020-01-08:
Hi,
I am using the BQ25896 library with Arduino and it works well with AVRs, but it isn't working with my BLE. This is a log:
```
In file included from C:\Users\Giorgio\AppData\Local\Temp\arduino_modified_sketch_145569\read_properties.ino:3:0:

C:\Users\Giorgio\AppData\Local\Temp\arduino_build_584731\sketch\BQ25896.h: In instantiation of 'EmbeddedDevices::BQ25896<CELL>::BQ25896(TwoWire&) [with int CELL = 1; TwoWire = arduino::MbedI2C]':

C:\Users\Giorgio\AppData\Local\Temp\arduino_modified_sketch_145569\read_properties.ino:5:31:   required from here

C:\Users\Giorgio\AppData\Local\Temp\arduino_build_584731\sketch\BQ25896.h:258:42: error: no matching function for call to 'arduino::MbedI2C::MbedI2C()'

             BQ25896(TwoWire& w) : wire(&w){};

                                          ^
```
I think that problem is similar in [issue#27](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/27) but I don't understand on how to fix the library
Can someone help me?

facchinm, 2020-01-09:
Hi @morettigiorgio ,
you can simply remove the "derivation" (no need to derive from TwoWire since all the functions are public).
Here's the patch, please close the issue if it is solved :wink: 
```patch
 diff --git a/BQ25896.h b/BQ25896.h
index 6123e6b..a98b334 100644
--- a/BQ25896.h
+++ b/BQ25896.h
@@ -13,7 +13,7 @@ namespace EmbeddedDevices
 {
     
     template <int CELL>
-    class BQ25896 : public TwoWire 
+    class BQ25896 
     {
         public :
             enum class TS_RANK
```

morettigiorgio, 2020-01-10:
Thanks Martino, now I can compile it.
Thank you so much again.

closed 2020-01-10 by morettigiorgio

morettigiorgio, 2020-01-10:
I have a notice for someone has a problem with its Arduino 33 BLE and this library (BQ25896) around the Wire(): i finded another problem on **Wire.endTransmission()** that to hang my sketch.
I fix it putting a **wire->begin();** before **wire->endTransmission(stop);**