arduino/ArduinoCore-nRF528x-mbedos #45: Arduino Nano 33 BLE

cacodoxy, 2020-01-09:
I am trying to create a second Hardware UART Serial Port on my Arduino NANO33 BLE, but I keep getting the error: no matching function for call to 'arduino::UART::UART(PinName&, PinName&, PinName, PinName)'

I followed your instructions at https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/38
but to no avail. I am attaching the code I am using as well. 

#include "Arduino.h"
#include "wiring_private.h"
#include "HardwareSerial.h"
#include <Serial.h>

void setup() {
  UART mySerial(digitalPinToPinName(4), digitalPinToPinName(3), NC, NC);
}

...
When I move the UART outside the setup code, I get the same thing. Please help.

facchinm, 2020-01-10:
Hi @cacodoxy ,
you need to update the core version to 1.1.3 (the signature of `UART` constructor didn't accept RTS and CTS before, so we added them here https://github.com/arduino/ArduinoCore-nRF528x-mbedos/commit/87a622fd622bfe3bcbd4aca0234b3283585fe101).
Please reopen if it's not working for you., thanks!

closed 2020-01-10 by facchinm

Technorico, 2020-01-16:
Hi, I would like to use a second uart, too. 
But when I type "UART" on the Arduino IDE, it isn't recognized by the compilator.
Furthermore I'm not able to find header files named wiring_private.h and Arduino.h.
Thanks in advance for the response. 

PS: Sorry for bad English

facchinm, 2020-01-16:
@Technorico make sure you select `Nano 33 BLE` from boards menu before compiling. This core doesn't provide any `wiring_private.h` that, hence the name, is private and shouldn't be used for any user facing code.