arduino/ArduinoCore-nRF528x-mbedos #49: error: redefinition of 'struct stat'

bergstar, 2020-02-08:
**Scetch**:
```
#include <sys/stat.h>
```

**Target**:
```
Arduino Nano 33 BLE
```

**Dump**:
```
Arduino: 1.8.11 (Windows 10), Board: "Arduino Nano 33 BLE"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Arduino\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Arduino\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Arduino\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -ide-version=10811 -build-path C:\Users\Arduino\AppData\Local\Temp\arduino_build_861866 -warnings=none -build-cache C:\Users\Arduino\AppData\Local\Temp\arduino_cache_622342 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.bossac.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\Arduino\Desktop\sketch_feb08a\sketch_feb08a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Arduino\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Arduino\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Arduino\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -ide-version=10811 -build-path C:\Users\Arduino\AppData\Local\Temp\arduino_build_861866 -warnings=none -build-cache C:\Users\Arduino\AppData\Local\Temp\arduino_cache_622342 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.bossac.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\Arduino\Desktop\sketch_feb08a\sketch_feb08a.ino
Using board 'nano33ble' from platform in folder: C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4
Using core 'arduino' from platform in folder: C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4
Detecting libraries used...
"C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10811 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\Arduino\\AppData\\Local\\Temp\\arduino_build_861866\\sketch\\sketch_feb08a.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10811 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\Arduino\\AppData\\Local\\Temp\\arduino_build_861866\\sketch\\sketch_feb08a.ino.cpp" -o "C:\\Users\\Arduino\\AppData\\Local\\Temp\\arduino_build_861866\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Arduino\\AppData\\Local\\Temp\\arduino_build_861866\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -DARDUINO=10811 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\Arduino\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\Arduino\\AppData\\Local\\Temp\\arduino_build_861866\\sketch\\sketch_feb08a.ino.cpp" -o "C:\\Users\\Arduino\\AppData\\Local\\Temp\\arduino_build_861866\\sketch\\sketch_feb08a.ino.cpp.o"
In file included from C:\Users\Arduino\Desktop\sketch_feb08a\sketch_feb08a.ino:1:
c:\users\Arduino\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\9-2019q4\arm-none-eabi\include\sys\stat.h:27:8: error: redefinition of 'struct stat'
   27 | struct stat
      |        ^~~~
In file included from C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_error.h:21,
                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Queue.h:28,
                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Mail.h:28,
                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/rtos.h:33,
                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/mbed.h:23,
                 from C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Arduino.h:32,
                 from C:\Users\Arduino\AppData\Local\Temp\arduino_build_861866\sketch\sketch_feb08a.ino.cpp:1:
C:\Users\Arduino\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_retarget.h:500:8: note: previous definition of 'struct stat'
  500 | struct stat {
      |        ^~~~
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```
I did also try to update arm-none-eabi-gcc from 7-2017q4 to 9-2019q4 - same results.

This bug is also related to:
[](https://github.com/adafruit/Adafruit_ImageReader/issues/26)

facchinm, 2020-02-10:
`mbed_retarget.h` redefines some toolchain's libc definitions to be platform independent.
I'm investigating if there's a way to avoid that file to always be included (something like HAS_STAT_H definitions)

bergstar, 2020-02-13:
@facchinm Thank you for looking into it. If I can assist with testing or any other light duty tasks - please let me know.

bergstar, 2020-03-08:
Just wanted to check if there is any updates?

facchinm, 2020-03-10:
PR created https://github.com/ARMmbed/mbed-os/pull/12607, thanks for the heads up!

sabas1080, 2020-07-11:
is this available in the current version? @facchinm

facchinm, 2020-07-30:
Closing as wontfix (on this repo) since we decided to keep mbed 5.x as base here (and the patch was merged in 6.x). For anyone needing this functionality, please use https://github.com/arduino/ArduinoCore-mbed (which also supports Nano33BLE)

closed 2020-07-30 by facchinm