arduino/ArduinoCore-nRF528x-mbedos #50: 1.1.4 breaks on Windows 10

dcuartielles, 2020-02-08:
We have tested blink on Nano 33 BLE Sense in two different computers with same results ... we cannot compile and get the following output:

Arduino: 1.8.11 (Windows Store 1.8.28.0) (Windows 10), Board: "Arduino Nano 33 BLE"

C:\Users\xyz\AppData\Local\Temp\arduino_build_866518/core\core.a(Serial.cpp.o): In function `operator()':

C:\Users\xyz\OneDrive\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Serial.cpp:91: undefined reference to `yield'

C:\Users\xyz\AppData\Local\Temp\arduino_build_866518/core\core.a(Serial.cpp.o): In function `arduino::UART::write(unsigned char const*, unsigned int)':

C:\Users\xyz\OneDrive\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Serial.cpp:86: undefined reference to `yield'

C:\Users\xyz\AppData\Local\Temp\arduino_build_866518/core\core.a(USBSerial.cpp.o): In function `waitForPortClose()':

C:\Users\xyz\OneDrive\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino\USB/USBSerial.cpp:26: undefined reference to `millis'

C:\Users\xyz\OneDrive\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino\USB/USBSerial.cpp:30: undefined reference to `delay'

C:\Users\xyz\OneDrive\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino\USB/USBSerial.cpp:28: undefined reference to `millis'

C:\Users\xyz\OneDrive\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\variants\ARDUINO_NANO33BLE/libs/libmbed.a(mbed_boot_gcc_arm.o): In function `__wrap_main':

/tmp/mbed-os-program/./mbed-os/rtos/source/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:77: undefined reference to `main'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano 33 BLE.

facchinm, 2020-03-11:
@dcuartielles have you been able to fix it (with IDE 1.8.12) or is still valid?

Rajmohankalapala, 2020-11-18:
IT IS INFACT HAPPENING. THANK YOU FOR MENTIONING.