arduino/ArduinoCore-nRF528x-mbedos #52: tone() not working as expected

theelims, 2020-02-22:
The function tone() is not working as expected. It outputs only a tone when the function is called with the duration argument. If I try to call tone() non-blocking without a duration no tone is audible.

morettigiorgio, 2020-02-24:
You try to use this code, in my Arduino BLE Sense it is working:
```
class Tone {
    mbed::DigitalOut   *pin;
    mbed::Timer        timer;
    mbed::Timeout      timeout;  // calls a callback once when a timeout expires
    mbed::Ticker       ticker;   // calls a callback repeatedly with a timeout
    uint32_t           frequency;
    uint32_t           duration;

public:
    Tone(PinName _pin, unsigned int frequency, unsigned long duration) : frequency(frequency), duration(duration)  {
      pin = new mbed::DigitalOut(_pin);
    }

    ~Tone() {
        stop();
        timeout.detach();
        delete pin;
    }

    void start(void) {
        ticker.attach(mbed::callback(this, &Tone::toggle), 0.5f / float(frequency));
        start_timeout();
    }

    void toggle() {
      *pin = !*pin;
    }

    void stop(void) {
        ticker.detach();
        pin = 0;
    }

    void start_timeout(void) {
        timeout.attach(mbed::callback(this, &Tone::stop), duration/1000.0f);
    }
};

Tone* active_tone = NULL;

void tone(uint8_t pin, unsigned int frequency, unsigned long duration) {
  if (active_tone) {
    delete active_tone;
  }
  Tone* t = new Tone(digitalPinToPinName(pin), frequency, duration);
  t->start();
  active_tone = t;
};

void noTone(uint8_t pin) {
  if (active_tone) {
    active_tone->stop();
    delete active_tone;
    active_tone = NULL;
  }
};
```

then you use this:
```tone(yourPin,4000,200);     // freq 4000Hz for 0.2s (200ms)```

what is your circuit? you try this: [https://easyeda.com/editor#id=|091dd9bc729749a4ad182e298a340be0](url)

theelims, 2020-02-25:
I'm using a MOSFET instead of an transistor but otherwise the same schematic.

Your code is basically what is already implemented in the tone.cpp.

Contrary to what is written on the Arduino Website this implementation of tone() is non-blocking. For now I will do a workaround with a very long duration. However, I would appreciate to have a version of tone() that plays indefinitely.

facchinm, 2020-03-11:
Hi @theelims ,
I created [a PR](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/59) to target your issue (you are right , the specifications request an infinite tone if duration is 0).
May you test it and report if it's working for you? Thanks

closed 2020-04-07 by facchinm

clwillingham, 2020-05-24:
Is the PR in production right now? because i'm currently working with the latest stable version through Arduino IDE and having this exact same issue. I can get a tone if I provide a duration, but no tone If i handle timing without a duration

specifically, this does not result in a tone:
```C
void setup() {
  pinMode(9, OUTPUT);
}

//expected: (500hz tone oscillating on and off at half second intervals
//result: no tone
void loop() {
  tone(9, 500);
  delay(500);
  noTone(9);
  delay(500);
}
```
but this does result in a tone
```C
void setup() {
  pinMode(9, OUTPUT);
}

//expected: 500hz tone oscillating on and off at half second intervals
//result: 500hz tone oscillating on and off at half second intervals
void loop() {
  tone(9, 500, 500);
  delay(1000);
}
```

facchinm, 2020-05-25:
@clwillingham the patch is not yet in production, we'll go live as soon as https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/31 is correctly fixed too