arduino/ArduinoCore-nRF528x-mbedos #55: Generating defines: No such file or directory

bissonex, 2020-02-24:
Hi, I am trying to generate a new variant:

_$ mbed-os-to-arduino EFM32GG_STK3700:EFM32GG_STK3700_

VARIANT=EFM32GG_STK3700 BOARD=EFM32GG_STK3700
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/private/tmp/mbed-os-program" (program)
[mbed] EFM32GG_STK3700 now set as default target in program "mbed-os-program"
[mbed] Working path "/private/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
Compiling Mbed Application...

Generating defines..../mbed-os-to-arduino: line 129: Arduino/hardware/arduino/mbed/variants/EFM32GG_STK3700/defines.txt: No such file or directory

manchoz, 2020-02-25:
Hi,
it seems that you are missing the `Arduino/hardware/arduino/mbed/variants/EFM32GG_STK3700` path/directory.

In order to understand more, you can run the tool with the Bash 'debug' option active:

```bash
    $ bash -x ./mbed-os-to-arduino EFM32GG_STK3700:EFM32GG_STK3700
```

Please, let me know.

facchinm, 2020-02-25:
Right now the script doesn't create `hardware/arduino/mbed/variants/$yourvariantname` which must be pre existent. Pull requests are well accepted but we are probably going to deprecate the "standard" profile very soon (so the variant folder with a real config should exist before running the script)

bissonex, 2020-02-25:
Yes indeed, that was the issue (missing variant folder). Any good docs for the whole process of adding a new target to Arduino IDE? I am new to this and would like to add the Silicon Labs STK3700 (https://os.mbed.com/platforms/EFM32-Giant-Gecko/) to the list of supported boards.

closed 2020-02-25 by bissonex

per1234, 2020-02-25:
@bissonex there is some documentation here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

and if you want to add Boards Manager installation support:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification

If you have any questions, feel free to post on the [Arduino Forum](https://forum.arduino.cc/) and we'll provide assistance.