arduino/ArduinoCore-nRF528x-mbedos #60: Compilation Path Length Issues with Dependant Tools

psytribalogical, 2020-03-17:
We have encountered an issue in using the toolchain included with this package ([full thread here](https://www.visualmicro.com/forums/YaBB.pl?num=1583720676/25#25)), whereby having too many libraries or paths in the compilation will fail with the error "arm-none-eabi-g++: error: CreateProcess: No such file or directory".

We have found that using a newer version of the arm-none-eabi toolset resolves this, and the one currently referenced in the packageindex.json is 2017-q4, which suffers from this issue, whereas updating to the STM32 9.2.1 version resolves this (from [this toolchain](https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v9.2.1-1.1/xpack-arm-none-eabi-gcc-9.2.1-1.1-win32-x32.zip), within this [board package](https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json))

Is there a plan to update the dependency at present? (or should this be logged elsewhere?)

An example of a failed and working compilation command are attached:-
[ArduinoBLE_FailedCompilationLengthIssue.txt](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/4344512/ArduinoBLE_FailedCompilationLengthIssue.txt)

facchinm, 2020-03-18:
Hi @psytribalogical ,
which WIndows version are you using?
AFAIK the commandline limit was more or less 32000 characters (hence this patch https://github.com/arduino/arduino-builder/commit/ea2e91bb2afb4abee66aea09bca0c5eda0d5dbe8).

psytribalogical, 2020-03-18:
This is on Win 10 Pro Version	10.0.18362 Build 18362, and that was the first thing I thought it was too (one of the imposed limits).

Thanks for the pointer to the builder elements on this, and the issue isn't the windows command processor, but the actual exe's internal processing (the newer tool chain works fine with longer commands via the same processor, and the error is being reported from within the arm-none-eabi exe itself).

An example sketch and  libraries are attached, along with compilation log, and in this example the inclusion of too many libraries manifests the issue way before the 32k limit (and if we remove the escaping from the build output the same length limitation is seen)

[Arduino_1-8-12_BuildOutput.txt](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/4350744/Arduino_1-8-12_BuildOutput.txt)
[DummyLibraries.zip](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/4350742/DummyLibraries.zip)
[SketchExample.zip](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/4350743/SketchExample.zip)

facchinm, 2020-03-23:
Updating to avr-gcc 9 could be a good idea but we need to evaluate the impact first on smaller targets (after merging https://github.com/arduino/actions/pull/18 everything should be easier)