arduino/ArduinoCore-nRF528x-mbedos #61: Arduino Nano 33 BLE - Pull-down resistor not working as intended

Talabo, 2020-03-18:
Hi,
I've been testing an Arduino Nano 33 BLE to test it's functionality with pull-down resistors. 

```
int testPin = 2;

void setup() {
  Serial.begin(9600);
  pinMode(LEDG, OUTPUT);
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(testPin, INPUT_PULLDOWN);
}

void loop() {
    if(digitalRead(testPin)){
      digitalWrite(LED_BUILTIN, HIGH);
      digitalWrite(LEDG, HIGH);
    }else{
      digitalWrite(LED_BUILTIN, LOW);
      digitalWrite(LEDG, LOW);
    }
    Serial.println(digitalRead(testPin));
}
```
A switch is hooked up between 3V3 and pin D2. With this code the micro controller is acting as if it's a pull-up resistor and not pulldown. The RGB-led seems also to not handle this code without adding a minimum of delay(1);

NB! another point worth mentioning. The RGB-led is fixed to 3V3, which means the code for "LOW" and "HIGH" is inverted.

facchinm, 2020-03-18:
Hi @Talabo ,
the issue gets solved by https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/31 ; I still didn't have time to properly finish the PR but if you could test it it would help the merge.

Talabo, 2020-03-19:
Hi @facchinm ,
pardon my lack of experience. Do you have a basic step-by-step on how to try out the #31 commit? Is it possible with arduino IDE or other IDEs?

facchinm, 2020-03-20:
The easiest way is probably downloading the core at that specific revision as zip (https://github.com/arduino/ArduinoCore-nRF528x-mbedos/archive/c_scoped_apis.zip) and replace the content of `C:\Users\yourUsername\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4` with the content of the zip :slightly_smiling_face:

Talabo, 2020-03-27:
> Hi @Talabo ,
> the issue gets solved by #31 ; I still didn't have time to properly finish the PR but if you could test it it would help the merge.

Hi, @facchinm ,

I confirm the code is working correctly with PULLDOWN resistor. The RGB-led still is inverted compared to HIGH and LOW, but it's a minor detail.

Code works good without need of a delay(1); aswell!

Measured around 13k ohm when pulled down. and 3,3v difference between the pin and 3v3 pin. Which confirms the functionality fixes #31 !! and which also fixes #30 ?

closed 2020-07-30 by facchinm