arduino/ArduinoCore-nRF528x-mbedos #62: USB HID - Keyboard being interpreted as UK layout

codekrafter, 2020-03-22:
I am currently writing an application that uses the USB HID library, and certain keys are being interpreted as if it is a UK keyboard instead of a US keyboard (0x40 is " instead of @). I have looked through code and not found anything that could be used to fix it (I have tried changing my vendor and product code to a microsoft keyboard but that did not change it). Any help would be appreciated!

facchinm, 2020-03-23:
Hi @codekrafter ,
the keyboard layouts unfortunately only make sense in conjunction with an operating system configured with the same locale. The file you need to modify to get a US keyboard is this https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/libraries/USBHID/USBKeyboard.cpp#L33 (simply add `#define US_KEYBOARD` at the top of the file and it should switch to US layout)

closed 2020-03-23 by facchinm

codekrafter, 2020-03-23:
I modified that file from the boards manager installation, but it did not properly change it. Are there other steps I need to do to properly recompile the library?

facchinm, 2020-03-24:
@codekrafter no other step should be needed. One thing you can try is inserting a compile error (like bogus strings) in the file at various places to make sure you are modifying the right file (and that the define is applied)

codekrafter, 2020-03-24:
I tried adding an #error direction and even a #warning directive to the header, but none of those seemed to be applied. I am also not able to install the hardware package by following the instructions in the repository.

facchinm, 2020-03-25:
@codekrafter are you using the Windows Store version of the IDE? Because in that case the folders being used are different (the default sketchbook should always be `C:\Users\yourUsername\Documents\Arduino` though).
If you run a compile with verbose turned on (from Preferences) the builder prints all the paths being used and it can help you modifying the right file.

codekrafter, 2020-04-04:
@facchinm I ran it with verbose on, and it is never detecting the usage of the USBHID library (or compiling those files), it might just be using the files builtin to the core library for the Arduino Nano 33 BLE

facchinm, 2020-04-06:
@codekrafter I may have missed go give you a piece of information... Sorry for that. In the sketch you should include "PluggableUSBHID.h" before any other include, otherwise the library being used will be mbed's (and the only way to change the layout would be recompiling libmbed).
In case everything is setup correctly you should get a print like:
```
Using library USBHID in folder: /run/media/martino/work/.arduino15/packages/arduino/hardware/mbed/1.0.6/libraries/USBHID (legacy)
```