arduino/ArduinoCore-nRF528x-mbedos #64: Is it possible to put the Arduino 33 BLE into bootloader mode from code?

morettigiorgio, 2020-04-07:
I would like to be sure that my Arduino is in upload mode before pressing the upload button from IDE. This is because my Arduino is not easy to reach and I often have to press the reset button to load the sketch.
I could put Arduino in upload mode, from bluetooth, with a code if this were possible.
Thanks

facchinm, 2020-04-07:
Hi Giorgio, sure you can :slightly_smiling_face: 
The function we invoke to reboot into bootloader is `_ontouch1200bps_()` (from [here](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/variants/ARDUINO_NANO33BLE/variant.cpp#L120)) .
You can add a BLE characteristic that will call that function when written.
I'm a bit more interested in why you often need to double press the reset button... Are you on Linux? In that case there might be `modemmanager` interfering with the upload procedure (as reported multiple times, for example here https://github.com/arduino/Arduino/issues/7690)

morettigiorgio, 2020-04-08:
Hi Martino.
It is perfect. It is properly that I looked for. Thanks
I hope I'm not off topic too:
The problem is only when I use the WDT routine.
At the end of compile, my IDE looks for COMs and Arduino changes from COM12 in COM13, the upload starting but after a few seconds, Arduino disconnects from USB (LED_BUILTIN flash 2 fast and 2 slow).
Only reset with bootload mode resolves it
It seems that WDT interrupts the uploading

facchinm, 2020-04-08:
Ah, it's what surely happens... How often is your watchdog configured to fire? From what I read on Nordic forum, there's no way to deactivate it using a soft reset, only the reset button works, so the code I gave you is unlikely to fix the problem.
A workaround could be to configure the watchdog period to a long one just before calling `NVIC_SystemReset` inside `_ontouch1200bps_()` (https://devzone.nordicsemi.com/f/nordic-q-a/53904/nrf52840-watchdog-for-arduino-nano-33-ble-sense/217952#217952) and then, when the sketch boots, reconfigure it to the desired period

morettigiorgio, 2020-04-08:
In fact, your code works only with WDT disable and my uploading problem is the same of your link indicates. I use that WDT code (5 sec) and once it started there is not way to stop it. I also tried to modify the WDT time with NRF_WDT->CRV = 3932159; (120s) before _ontouch1200bps_(); but, unfortunately, it seams irrelevat

leomonan, 2020-04-17:
I use this code and it works (reset to bootloader):

```
NRF_POWER->GPREGRET = 0xB0UL;
NVIC_SystemReset();
```

manchoz, 2020-04-17:
@morettigiorgio in order to actually change the timeout value you also need to feed the WDT. Please, refer to [Reload Criteria](https://infocenter.nordicsemi.com/topic/ps_nrf52840/wdt.html?cp=4_0_0_5_35_0#concept_d5y_nzw_sr) for more information.

The complete sequence should be:

```C++
constexpr uint32_t WDT_MAX_VALUE = 0x07CFFFFF;
NRF_WDT->CRV = WDT_MAX_VALUE;
for (auto rr = 0; r < 8; r++)
  NRF_WDT->RR[rr] = NRF_WDT_RR_VALUE;
```
This feature is also available via the NRF5 SDK that you can use directly from the Arduino sketch.

```C++
constexpr uint32_t WDT_MAX_VALUE = 0x07CFFFFF;
nrf_wdt_reload_value_set(WDT_MAX_VALUE);
nrfx_wdt_feed();
```

FYI, NRF52840's watchdog is fully supported via the [MbedOS Watchdog API](https://os.mbed.com/docs/mbed-os/v5.15/apis/watchdog.html) even on Arduino Nano 33. Still, the MbedOS Watchdog API is pretty simple/naive and **Nordic's Watchdog can't be stopped**.

morettigiorgio, 2020-04-17:
Thanks @leomonan and @manchoz, I will test them soon.

facchinm, 2020-04-22:
Closing as answered :wink:

closed 2020-04-22 by facchinm