arduino/ArduinoCore-nRF528x-mbedos #65: Arduino Nano BLE 33, low voltage from digitalWrite()

GitLorenzo, 2020-04-16:
Hello,

I am having issues with the easiest code ever:

`void setup() {
  pinMode(8, OUTPUT);
}

void loop() {
   digitalWrite(8, HIGH);
}`

The problem is that the voltage I get from the pin 8 is not 3.3, but lower and not enough to fully turn on a LED or make an active buzzer ring.
I know this is an issue somebody brought up in the forum (https://forum.arduino.cc/index.php?topic=663776.0), but could not find any solution around.
Is this something you can help me with?

Thank you very much in advance!
ciao,
Lorenzo

morettigiorgio, 2020-04-16:
For a buzzer circuit, you can try this:
https://easyeda.com/editor#id=091dd9bc729749a4ad182e298a340be0

facchinm, 2020-04-17:
Hi @GitLorenzo ,
we are aware of the issue, and it should get solved by this patch https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/31 (that I'm reworking, hence it's not yet merged).
The problem is that the pin get reinitialized every time in the loop, and the default value is LOW, so it likely "rings" when observed with a scope.

GitLorenzo, 2020-04-17:
Thank you both for your help!
Using a transistor I can get both led and active buzzer working as Giorgio suggested!
But I would like to implement the patch #31 as well. One stupid question since I am definitely not an expert. How should I do that? Do I have to add the green lines and remove the red lines from each file? Or, is there a different way to do that?

closed 2020-07-30 by facchinm