arduino/ArduinoCore-nRF528x-mbedos #66: Can I build libmbed.a myself?

leomonan, 2020-04-17:
I want to add some debug code inside “class Thread”, but I find this class may be implemented in 
> ARDUINO_NANO33BLE/libs/libmbed.a
 Can I find a way to get the libmbed.a's source code and build it locally?

facchinm, 2020-04-17:
Hi @leomonan ,
you can just run `./mbed-os-to-arduino -r /path/of/you/mbed-os/local/clone ARDUINO_NANO33BLE:ARDUINO_NANO33BLE` from the root of this repository.
The script runs on Linux and probably on macOS (you may need to apply [this](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/57) patch).
I just noticed that the instructions in README are out of date and I'm going to amend them.

closed 2020-04-17 by facchinm

leomonan, 2020-05-28:
Hi 
   I got "mbed: command not found" error, I can't find mbed function in this script.
> $ ./mbed-os-to-arduino -r local/clone ARDUINO_NANO33BLE:ARDUINO_NANO33BLE
> MBED_CLEAN=0
> MBED_UPDATE=0
> APPLY_PATCHES=0
> LOCAL_REPO=local/clone
> REMOTE_BRANCH=
> MBED_CORE_LOCATION=/home/monan/workspaces/ArduinoCore-nRF528x-mbedos
> 
> VARIANT=ARDUINO_NANO33BLE BOARD=ARDUINO_NANO33BLE
> Checking for prerequisites... done.
> Creating MbedOS Application..../mbed-os-to-arduino: line 25: mbed: command not found

facchinm, 2020-05-28:
You need to install `mbed` cli as a prerequisite (https://os.mbed.com/docs/mbed-os/v5.15/tools/manual-installation.html) , I need to amend the documentation to point it out.

trungnq97, 2020-08-10:
Hi @facchinm 
I run the below command in `/workspace/.platformio/packages/framework-arduino-nrf52-mbedos`

`./mbed-os-to-arduino ARDUINO_NANO33BLE:ARDUINO_NANO33BLE`

and get the error below. Do you know what could be the issue? Thank you in advance.

```
MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=0
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/workspace/.platformio/packages/framework-arduino-nrf52-mbedos

VARIANT=ARDUINO_NANO33BLE BOARD=ARDUINO_NANO33BLE
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] ARDUINO_NANO33BLE now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Compiling Mbed Application...Compile [ 49.1%]: thread_management_api.c
[mbed] ERROR: "/home/trung/.platformio/penv/bin/python" returned error.
       Code: 1
       Path: "/tmp/mbed-os-program"
       Command: "/home/trung/.platformio/penv/bin/python -u /tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m ARDUINO_NANO33BLE --source . --source ./mbed-os/features/unsupported/dsp/ --build ./BUILD/ARDUINO_NANO33BLE/GCC_ARM -v"
```

leomonan, 2020-09-21:
@trungnq97 this my caused by a compile error(did you modified the source code youself?), you can find the error by run the build command separately:
/home/trung/.platformio/penv/bin/python -u /tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m ARDUINO_NANO33BLE --source . --source ./mbed-os/features/unsupported/dsp/ --build ./BUILD/ARDUINO_NANO33BLE/GCC_ARM -v