arduino/ArduinoCore-nRF528x-mbedos #67: How do I use the mbed::nfc feature

Gerriko, 2020-04-25:
Sorry these aren't really issues, but questions related to the NFC-A Tag feature. I put this question on the Arduino.cc forum but in case not everyone looks at that forum I wanted to ask the same question here.

I wish to use the nfc feature with my nano ble 33 board and I see that within ArduinoCore-nRF528x-mbedos... /arduino/mbed/features folder there is a folder for nfc.

So, is this code ready for use (it looks a lot less code than say the Arduino Primo nfc library, which is also nRF52 based)? 

If so, how to I use this code? 

Then on the reverse side of the board are two solder pads for P0.09 and P0.10 which you can join to D7 and D8.

Is there any way you can use P0.09 and P0.10 directly in our firmware? If so, what is the coding format?

facchinm, 2020-04-27:
Hi @Gerriko ,
AFAIK only PN512 nfc chips are supported on mbed-os out of the box https://github.com/ARMmbed/mbed-os/tree/master/features/nfc/source/controllers . The IP contained in the nRF can only act as passive NFC, not as a reader. You can still use the APIs from Nordic  SDK since they are included in the build (some examples and reference [here](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v15.0.0/examples_nfc.html))
As for the pins, your analysis is correct: the two solder pads allow you to use D7 and D8 to connect the antenna, otherwise if you want to use that pins as digital you can directly solder the antenna to the exposed pads.

Gerriko, 2020-04-27:
Hi @facchinm 

Thanks for your feedback. 

Yes, I eventually discovered the mbed documentation, which describes the MBED NFC API and where the PN512 module fits. https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/nfc/nfc_design.md

There may be ways to incorporate parts of this mbed library, such as the ndef module, but this looks like a good deal of work, especially if you are not familiar with this mbed library. 

Anyhow, I believe a Chiara Ruggeri (from Arduino) has already developed some great nRF52840 NFC examples for the Arduino Primo boards. Is there any simple way of making these examples work with the nano BLE 33? The issue for me is that those examples were created as core examples for the primo so are all residing in the Arduino15 folder.

Gerriko, 2020-04-28:
Not sure if this should be raised here or on the Arduino forum.

Started playing with the Nordic nRF52 nfc code examples and began with a very simple sketch which will turn on LED when NFC field detected (code not complete). However this threw up an error:

...\Arduino\libraries\nRF52_NFC\src\hal_t2t\hal_nfc_t2t.c:39:10: fatal error: nrf_drv_clock.h: No such file or directory

 #include "nrf_drv_clock.h"

Is this something I can fix?

==========================
MY UNFINISHED SKETCH:
`#include <stdint.h>
#include "nfc_t2t_lib.h"
static void nfc_callback(void * p_context, nfc_t2t_event_t event, const uint8_t * p_data, size_t data_length)
{
    (void)p_context;
    switch (event)
    {
        case NFC_T2T_EVENT_FIELD_ON:
            digitalWrite(LED_BUILTIN, HIGH);
            break;
        case NFC_T2T_EVENT_FIELD_OFF:
            digitalWrite(LED_BUILTIN, LOW);
            break;
        default:
            break;
    }
}
void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);
 // add callback etc....
}
void loop() { }
`

Gerriko, 2020-04-28:
Which version of the nRF52840 SDK is being used. I am trying to port across some the nRF52840 NFC examples, as suggested, and picked up on a "hal" discrepancy. My program was looking for a particular "nrf_" file which exists in the latest nRF52840 SDK (v16) but not found on my local computer in the Arduino15 folder.

Gerriko, 2020-04-29:
This porting across the Nordic SDK NFC examples is proving quite messy. I have discovered that the Arduino Primo examples must be using an old Nordic SDK as many of those functions have changed. So not much use for the new nano BLE 33 board.

The question I have now is how do I deal with the Nordic SDK "nrfx" module (i.e. the nrfx folder with all subfolders and associated nrfx files). Is this not something that is part of the nano BLE core or do I assume this to be separate and thus can block copy as part of a library.

omega1119, 2021-01-12:
Hi there,

I have been looking at the following two examples:

https://os.mbed.com/docs/mbed-os/v6.5/apis/nfc-apis.html
https://os.mbed.com/docs/mbed-os/v6.5/apis/nfc-eeprom.html

In the first example I am able to create the message, however I'm struggling with writing the message to memory.

The second example has this code

`mbed::nfc::vendor::ST::M24srDriver m24sr_driver(D14, D15); 
events::EventQueue queue; 
mbed::nfc::NFCEEPROM nfc(&m24sr_driver, &queue, ndef_buffer); 
nfc.write_ndef_message();`

The issue I am having is how do I get the driver for the Arduino Nano 33 BLE Sense? 

This line

`mbed::nfc::vendor::ST::M24srDriver m24sr_driver(D14, D15);`

I can't seem to find any good examples of how this would work. Any help would be greatly appreciated. In terms of what I am trying to achieve is to set the Arduino up using an NFC antenna as a tag so I can read out a message when I place an iPhone near it for example.

Thanks