arduino/ArduinoCore-nRF528x-mbedos #70: Watchdog Timer

tjpetz, 2020-04-26:
Request you add support for the Watchdog timer in mbed.  The headers are there.  Support became available for the NRF52840 in Oct 2019 in mbed.  It looks like just a define DEVICE_WATCHDOG is likely required in defines.txt to enable.

Is there any reason (other than timing as this code and the WDT support for the NRFs came out around the same time) the Watchdog was excluded?

Thanks in advance.

facchinm, 2020-04-27:
Hi @tjpetz ,
no reason except that the core was compiled against a version of mbed-os that didn't contain that code. Since we are in sync with master, next rebuild will contain it.
Leaving the issue open to keep track of it.

tjpetz, 2020-04-28:
Thanks for the response.  I'll keep on the lookout for the next rebuild.