arduino/ArduinoCore-nRF528x-mbedos #76: analogWrite to LED_PWR and digitalWrite after analogWrite

FemmeVerbeek, 2020-05-18:
It looks like two bugs to me.

- analogWrite to LED_PWR hangs up the program
- digitalWrite to any pin stops working after being preceded by a call to analogWrite to the same pin. 

my not so beautiful workaround

void digitalWrite2(int pinnr , boolean pinhigh){
  if (pinnr==LED_PWR) digitalWrite(LED_PWR, pinhigh);
  else analogWrite(pinnr, 255*pinhigh);
}
void analogWrite2(int pinnr ,uint8_t value){
if (pinnr==LED_PWR) digitalWrite(pinnr,value>127);
else analogWrite(pinnr,value);
}

It would be better to solve it by redefining the functions itself but so that it still compiles for other boards.
- Is there an #ifdef compiler directive to check that we are compiling for this type of board.
- while redefining, how would you point back to the orginal funtions

facchinm, 2020-05-26:
Hi @FemmeVerbeek 
`analogWrite(LED_PWR, 10);` should just work in any condition unless you already have more than 4 other PWM active (I just tested it). The reason is this assert (https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/pwmout_api.c#L146) that checks how many PWM channels are already in use.
Is this your case?

FemmeVerbeek, 2020-05-26:
Hi Martino,
    Thx for looking into it. I was making art with the onboard leds.
      :-)
    
    It looks like analog calling the 5th channel is what caused the
      hanging problem.  
    
    Apparently assigning a PWM channel stops digitalWrite from
      working on that pin . 
    
    Is there a (user friendly) way to free up the Pin channel in a
    sketch, so that digitalWrite works again? Possibly reassign the
    channel to another pin?
    Kind regards
    Femme Verbeek.
    
    code
    void setup() 
      { pinMode(LED_BUILTIN,OUTPUT);
        analogWrite(LED_BUILTIN,75);
        digitalWrite(LED_BUILTIN,1);
        delay(1000);
        digitalWrite(LED_BUILTIN,0);
      }
      void loop(){}
    
    
    
    
    
    
    
    
    
    
    
    Op 26-5-2020 om 11:23 schreef Martino
      Facchin:
    
    
      
      Hi @FemmeVerbeek
        analogWrite(LED_PWR, 10); should just work in any
        condition unless you already have more than 4 other PWM active
        (I just tested it). The reason is this assert (https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/pwmout_api.c#L146)
        that checks how many PWM channels are already in use.
        Is this your case?
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/76#issuecomment-633913036",
"url": "https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/76#issuecomment-633913036",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]
    
Met vriendelijke groet
Kind regards

Femme Verbeek
+31 6 26060303
https://www.linkedin.com/in/femmeverbeek/

JeffAnton, 2020-05-28:
I've hit this problem also.  digitalWrite after an analogWrite to the same pin does not work.  Once I saw this report I changed my digitalWrite calls into analogWrite with values of 0 or 255 and my project is working properly.  BTW: I have four analogWrite pins in use.