arduino/ArduinoCore-nRF528x-mbedos #78: Please explain '-iwithprefixbefore' in 'includes.txt'

oclyke, 2020-05-20:
Hi @facchinm (or others 🙂) - Inspired by this project I am trying to build a similar [mbed-os core for the Apollo3](https://github.com/sparkfun/Arduino_Apollo3/tree/v2.0.0-from-scratch). 

I have faced an issue trying to supply additional include search paths to the compile steps in a portable (cross platform) capable way. 
* Using the ```@file``` syntax from mbed's generated include directly won't work because those paths are relative and I can't guarantee the working directory that my compiler is called from (can I?)
* I've tried writing scripts to use the IDE's ```{runtime.platform.path}``` variable to generate a list of search paths that are absolute, however this bloats the number of characters from ~16k (mbed generated include file) to ~40k (varies based on the exact install location of the core) which causes an error running GCC. (~32k characters [seems to be the limit for a single command line call](https://stackoverflow.com/questions/3205027/maximum-length-of-command-line-string))

At a loss I came back to this project for a reference. I see in your variant's ```includes.txt``` it has this leading ```-iwithprefixbefore/``` *and* that it is directly included in your compile / link lines (```"@{compiler.mbed.includes}"``` -> ```nano33ble.compiler.mbed.includes={build.variant.path}/includes.txt```)

Can you please explain what that prefix is doing? How is it handled by the Arduino IDE? 

Thanks a lot - I have been trying hard to determine a way to expose the full mbed API to the user in this core.

facchinm, 2020-05-22:
Hi @oclyke !
That procedure takes advantage of https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html , so that paths length is greatly reduced (and can be specified as relative to `{runtime.platform.path}` ).
I think the missing piece you are looking for is `"-iprefix{build.core.path}"` ; in this way, gcc replaces internally the `withprefixbefore` string with the one provided.

Glad to hear you are working on the Apollo3! Let me know how it proceeds :wink:

oclyke, 2020-05-26:
Thanks a bunch @facchinm! I was previously unaware of that option in GCC. It will be invaluable in the project and future work as well. You can always check out the Apollo3 project here: https://github.com/sparkfun/Arduino_Apollo3

The v2.0.0 release will be based on mbed so that we can support [ArduinoBLE](https://www.arduino.cc/en/Reference/ArduinoBLE) as directly as possible.

closed 2020-05-26 by oclyke