arduino/ArduinoCore-nRF528x-mbedos #82: Only seeing rising-edge interrupts.

pbaughman, 2020-06-11:
Hello!  I've got some code that I'm porting to the Arduino Nano BLE.  I'm having some difficulty because I think I'm only seeing rising edges in my interrupt handler instead of rising and falling edges.  The following code minimally reproduces the issue:

```
#define TEST_PIN 3

void setup() {
  Serial.begin(9600); // open the serial port at 9600 bps for debugging
  pinMode(TEST_PIN, INPUT_PULLUP); 
  attachInterrupt(digitalPinToInterrupt(TEST_PIN), enc_tick_a, CHANGE);
}

volatile bool int_a = false;
volatile uint8_t val_a = 0;

void enc_tick_a(){
  int_a = true;
  val_a = digitalRead(TEST_PIN);
}

void loop() {
  if (int_a) {
    Serial.println("Interrupt A");
    Serial.println(val_a);
    int_a = false;
  }
}
```

There's a rotary encoder attached to the pin under test.  When I run this code on `Arduino Duemilanove`, I see the following output for every "tick" of the encoder:

```
Interrupt A
0
Interrupt A
1
```

Continuing to turn the encoder, I see what one would expect:

```
Interrupt A
0
Interrupt A
1
Interrupt A
0
Interrupt A
1
Interrupt A
0
Interrupt A
1
```

I have an identical hardware setup but with an `Arduino Nano Every` chip that produces the same output.

When I test with the `Arduino Nano 33 BLE` chip I'm not seeing falling-edge interrupts.  The output from the above test program is only:

```
Interrupt A
1
```

Multiple ticks of the encoder yields

```
Interrupt A
1
Interrupt A
1
Interrupt A
1
Interrupt A
1
Interrupt A
1
Interrupt A
1
Interrupt A
1
```

As you can see, it appears my interrupt is only firing on 0->1 transitions but not 1->0 transitions.

If I change TEST_PIN to 2 (the other leg of the encoder) I observe the same thing.

Am I missing something fundamental here, or is `attachInterrupt(x, x, CHANGE)` not working as expected on the BLE nano?

Thank you!

pbaughman, 2020-06-11:
Some additional testing:

If I change `CHANGE` to `FALLING` in `attachInterrupt` I see:

```
Interrupt A
0
```

So it seems like the board is capable of detecting the falling-edge interrupts, its' just not doing so when the interrupt mode is CHANGE.

pbaughman, 2020-06-11:
One additional wrinkle - I see that there's a closed issue describing this same problem that has a merged MR: https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/68

I believe I'm using the latest `master` - I've followed the instructions in the README.md file.  Is there a way to tell for sure?

pbaughman, 2020-06-11:
Final update - I turned up the verbosity when building and saw that I was probably still using the installed version instead of the source version of `ArduinoCore-nRF528x-mbedos`  Upon discovering this, I uninstalled the version installed by the board manager.  Now my IDE detects the code installed in the `hardware` folder of my sketchbook.

Unfortunately trying to build results in the error `exec: "/bin/arm-none-eabi-g++": file does not exist`

Can someone confirm that the instructions in the README.md are for linux systems only?  Are they intended to work on Windows, too?  I suspect I'm not on the happy development path here and that's why I'm encountering problems.

pbaughman, 2020-06-11:
Alright, it was a real challenge to get this working on a Windows system - I will post the steps I followed in case they are helpful to someone else:

1) Make sure the BLE Nano board is installed using the Arduino IDE
2) In the folder C:\Users\Username\AppData\Local\Arduino15\packages\arduino\hardware rename the folder `mbed` to `mbed_bak`  This will prevent the IDE from using the installed sources for the BLE board, but still leave the tools installed in `C:\Users\Peter\AppData\Local\Arduino15\packages\arduino\tools`
3) In your Arduino sketch folder (C:\Users\Username\Documents\Arduino) on my system, follow the instructions in README.md to create a `hardware/arduino` folder containing mbed and the ArduinoCore-API source

Then, for some reason I also had to edit C:\Users\Username\Documents\Arduino\hardware\arduino\mbed\platform.txt and change

```
tools.bossac.path={runtime.tools.bossac-1.9.1-arduino1.path}
```

to 

```
tools.bossac.path=C:\Users\Peter\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1
```

Once you've made all of those change, make sure to **restart your IDE.**

You should now (hopefully) be able to build with the source version of ArduinoCore-nRF528x-mbedos

closed 2020-06-11 by pbaughman

facchinm, 2020-06-12:
Hi @pbaughman ,
sorry but I've seen it just now that you solved the issue by yourself :slightly_smiling_face: 
For future reference, you can have both the official and the git version of a core installed sideways.
What you need to do is to create a folder in your sketchbook called `hardware/$foo/` and clone the core there.
For example, for this core, given `C:\Users\M\Documents\Arduino` as sketchbook path, it would run like this:
```
cd C:\Users\M\Documents\Arduino
mkdir -p hardware\arduino-git
cd hardware\arduino-git
git clone https://github.com/arduino/ArduinoCore-nRF528x-mbedos mbed
#for this particular core you should copy the "api" folder from another installation since it's just a placeholder in git
```
Restart the IDE and you are done; this way there's no chance to lose your modification in case of update.