arduino/ArduinoCore-nRF528x-mbedos #85: SPI.transfer16 not working as expected

rmlearney-digicatapult, 2020-06-19:
One expects `SPI.transfer16()` to be functionally equivalent to 2x sequential calls to SPI.transfer().

However the following functions produce very different results on my Nano 33 BLE communicating with an ADS1118, despite identical appearance. In particular, the function using SPI.transfer16() does not work but the SPI.transfer() one does.

```c++
int16_t ADS1118::getData(void){
  uint8_t buffer[2];
  int16_t retVal;

  setConfig(ADS1118_WRITE_CONFIG);

  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE1));
  buffer[0] = SPI.transfer(((_config >> 8) & 0xFF));
  buffer[1] = SPI.transfer((_config & 0xFF));
  SPI.transfer(0x00); // Have to do 32-bit write if !CS held low
  SPI.transfer(0x00);
  SPI.endTransaction();

  retVal = ((buffer[0] << 8) | buffer[1]);
  return retVal;
}
```

```c++
int16_t ADS1118::getData16(void){
  int16_t retVal;

  setConfig(ADS1118_WRITE_CONFIG);

  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE1));
  retVal = SPI.transfer16(_config);
  SPI.transfer16(0x00); // Have to do 32-bit write if !CS held low
  SPI.endTransaction();

  return retVal;
}
```

facchinm, 2020-06-22:
Indeed, the current SPI `transfer16` implementation does not consider MSBFIRST or LSBFIRST defines, using the processor's builtin byte arrangement. I just pushed https://github.com/arduino/ArduinoCore-nRF528x-mbedos/pull/81/commits/92833a2e4f6fe5812b0c6e50074682544a5d2d3e to fix it, let me know if it works so it can be merged safely.

rmlearney-digicatapult, 2020-06-22:
@facchinm that looks like it fixes it, thank you!

rmlearney-digicatapult, 2020-06-26:
@facchinm I've been trying to think how this works in terms of endian-ness and could you maybe clarify for me?

You make a 16-bit word union from 2x 8 bits that goes `<16 15 ... 9 8> <7 6 ... 1 0>`

So if you send `MSBFIRST` it sends out bits in the following order:

`<16 15 ... 9 8> <7 6 ... 1 0>`

But if you send `LSBFIRST` it just reverses the order of these bytes, not the bits, and sends:

`<7 6 ... 1 0> <16 15 ... 9 8>`

So how does this actually solve the problem to make sure the SPI bus is transmitting MSB (bit 16) or LSB (bit 0) first?

closed 2020-07-03 by facchinm