arduino/ArduinoCore-nRF528x-mbedos #86: link error on mbed::InterruptIn

zvilius, 2020-06-23:
Using MbedPinMode parameters with the mbed::InterruptIn class results in link errors.

```
mbed::InterruptIn chan_a{PinName::P1_11, MbedPinMode::PullNone}; // #1
//mbed::DigitalIn chan_a{PinName::P1_11, MbedPinMode::PullNone}; // #2 is ok
//mbed::InterruptIn chan_a{PinName::P1_11};                      // #3 is ok by itself

void setup()
{
    //chan_a.mode(MbedPinMode::PullNone);       // this (in combination with #3) is bad
}

void loop()
{
}
```
The error (for case 1) is:
"C:\\Users\\Zest\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/libs/libcc_310_trng.a" -Wl,--no-whole-archive -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group

C:\Users\Zest\AppData\Local\Temp\arduino_build_748706\sketch\sketch_jun23b.ino.cpp.o: In function `__static_initialization_and_destruction_0':

C:\Users\Zest\Documents\Arduino\sketch_jun23b/sketch_jun23b.ino:1: **undefined reference to `mbed::InterruptIn::InterruptIn(PinName, MbedPinMode)'**

If I use the mbed::DigitalIn class (case 2) there is no error.
If I use the 1-argument constructor (case 3) there is no error.
However if I use the 1-argument constructor and then call the mode() method (see above), the mode() method fails to link.

Perhaps there is a build issue with the mbed library as it relates to the macro redefinition of the mbed version of PinMode to MbedPinMode? However it works for mbed::DigitalIn.

zvilius, 2020-06-23:
I just noticed issues #7 and #9 which are likely related.

zvilius, 2020-06-24:
Thank you @Polido, I'll give that a try.

zvilius, 2020-06-24:
Apologies, but I'm new to the Arduino IDE and infrastructure so I don't know how all the pieces fit together. Perhaps you can just get me headed in the right direction:

I'm not sure I'm building with the new (beta) Arduino core for mbed devices.

I followed the installations instructions from the README:
* Clone the repo into $sketchbook/hardware/arduino
* Check out the beta branch
* Clone the ArduinoCore-API repo elsewhere
* Check out the namespace_arduino branch
* Make the "api" symbolic link point to the right place

When I run the Arduino IDE, it does seem to be noticing the new board in the $sketchbook folder:

[image]

But when I build (replacing the old MbedPinMode with PinMode) I get an error:

[image]

The reason I don't think I'm using the new stuff is the lines:
"Using board 'nano33ble' from ..." and "Using core 'arduino" from ..." refer to the ArduinoData folder rather than Arduino (which is the sketchbook folder).

Thanks for any assistance...

facchinm, 2020-06-25:
Hi @zvilius , 
I think the problem might be due to the folder naming; to make things easier you can directly download mbed core 1.2.0 (search for `Portenta` in board manager) that also supports the nano33ble and contains the modifications from the beta branch.
That core is going to superseed this one in a few months so it will also be valuable testing :slightly_smiling_face:

zvilius, 2020-06-25:
Hi Martino @facchinm ,
I did as you suggested, and the sketch above does build. I do get a number of warnings related to deprecated timing-related usage: switch to using chrono. Should I write an issue on those?

I'll see if I can get some real code to run now.

Thanks.

facchinm, 2020-06-29:
Hi @zvilius ,
the warnings are the reason why the `mbed` core is still under `arduino-beta` organization :slightly_smiling_face: 
mbed 6 brought a lot of new deprecations / APIs removal that are not super stable but we are going to get that fixed soon.

zvilius, 2020-06-29:
Hi Martino @facchinm ,
I certainly understand. I was just asking whether I should write an issue or if that's not necessary. Sounds like it's not necessary...

facchinm, 2020-06-29:
Perfect, no need to open an issue, thanks :wink:

facchinm, 2020-07-30:
Closing as fixed

closed 2020-07-30 by facchinm