arduino/ArduinoCore-nRF528x-mbedos #88: digitalRead() pull-down not working

fgaetani, 2020-06-26:
Hi, I have noticed a serious issue on digitalRead () on any pin. I have a system with a 47k pull-down resistor, and a switch that pull the pin high. However, I noticed that if I don't run the digitalRead () function the pin remains low (measured with a multimeter), when I run the digitalRead () function the pin goes high (about 2.6 Volt) unless I use a very low pull-down resistance (about 4.7 k).

How can I solve it?

Thank you

facchinm, 2020-06-29:
Hi @fgaetani ,
the issue should be solved by `beta` branch (that I plan to release this week) as well as from upcoming `mbed` core (already released for PortentaH7 but it also contains support for the Nano33BLE).

Short explanation: the current porting layer for gpios wraps mbed's DigitalInOut objects in an ephemeral way; unfortunately, the mbed constructor that gets continuously called has a default pull (an internal PullUp in case of nRF52 platforms) that gets activated every time you create one of these objects.

The new layer only initializes the object once to fix this in a sane way.

fgaetani, 2020-06-29:
Hi @facchinm ,

thanks a lot for the info.
I solved momentarily by editing wiring_digital.cpp file. I have edited pinMode and digitalRead functions by including the code of the beta version (also modifying other linked files).
So will a new branch of this library come out or will we need to use ArduinoCore-mbed from now on?
I was not able to compile with ArduinoCore-mbed beta library because I had several errors from other libraries (e.g. ArduinoBLE).

Thank you

Best regards,

facchinm, 2020-06-29:
ArduinoBLE compatibility with ArduinoCore-mbed should be solved by this https://github.com/arduino-libraries/ArduinoBLE/commit/3b228dc90262751bdac97712a6c7bda95fcf3a5c (planning to release today :slightly_smiling_face:  )

closed 2020-06-29 by facchinm

fgaetani, 2020-06-29:
> ArduinoBLE compatibility with ArduinoCore-mbed should be solved by this [arduino-libraries/ArduinoBLE@3b228dc](https://github.com/arduino-libraries/ArduinoBLE/commit/3b228dc90262751bdac97712a6c7bda95fcf3a5c) (planning to release today 🙂 )

So the new reference to use will be ArduinoCore-mbed? Will ArduinoCore-nRF528x-mbedos be deprecated?

facchinm, 2020-06-29:
That's the plan, keeping `ArduinoCore-nRF528x-mbedos` on mbed 5.x (and making it obsolete) and moving to mbed 6 with `ArduinoCore-mbed`