arduino/ArduinoCore-nRF528x-mbedos #90: aurdino nana 33 ble serial connection no longer recognized by mac mojave, worked earlier today

AndySantaCruzAnalytics, 2020-07-03:
I can no longer upload my sckect to my ardunio nano 33 ble.

It worked fine early today. 

Not sure why but my mac no longer has a /dev/cu.usbmodem2140 ?

I was using screen to capture serial port output without any issue. serial port monitor worked fine

I tried rebooting my mac. Still have same problem

The board is marked ardunio cc . on it I do not think it is a clone

any suggestions would be greatly appreciated

AndySantaCruzAnalytics, 2020-07-05:
I do not know why but it seems to be working now.

I rebooted my mac several times. I do not think that fixed.

I switched to a working board, and uploaded blink

I pressed the reset button 2 time

and switch to the broken board

now it works

I tried to break it again by pressing the rest button 2 times.  still works?

I noticed that while the board was broken the green LED was solid and the orange lED blicked quickly

once it started working again it seems like the orange LED was blinking slowly?

facchinm, 2020-07-06:
Hi @AndySantaCruzAnalytics ,
if the board starts emitting the "morse-like" blinking pattern it means that something has gone very wrong (memory corruption or similar). Double clicking the reset button will bring the board in bootloader mode so you can always use that method as a failsafe.
To discover which bug caused the hang you can hook a serial-to-usb converter to pin TX1 (serial configured as 115200@8n1) and the system will report where the crash happened and what was the root cause.

closed 2020-07-06 by facchinm