arduino/ArduinoCore-nRF528x-mbedos #92: mbed os configuration

zvilius, 2020-07-25:
Could someone point me to where/how Mbed is configured when it is built in conjunction with ArduinoCore?

Some of the parameters I'd like to discover .. and possibly modify .. are:

1. Does Mbed use the PSP stack for threads? (I assume so, but would like to verify .. wouldn't want to change.)
2. What are the default sizes for the MSP and PSP stacks?
3. Is the FPU enabled by default? (I assume so .. perhaps I'll want to change for my application.)

Thanks,
Mark Z.

facchinm, 2020-07-27:
Hi @zvilius , 
the script we use to compile libmbed uses a mbed_app.json default configuration unless the core variant contains a proper configuration file (see https://github.com/arduino/ArduinoCore-mbed/blob/master/mbed-os-to-arduino#L74 ).
For example, this is the configuration for Portenta board (https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M7/conf/mbed_app.json).

* You can tune the stack size by setting `rtos.main-thread-stack-size` property.
* FPU is enabled by mbed on all supported processors (however, the build is `softfp`, not `hardfp`)

closed 2020-07-27 by facchinm

zvilius, 2020-07-29:
Thanks Martino @facchinm . That helps.