arduino/ArduinoCore-nRF528x-mbedos #93: Problem with Wire when rebuilding mbed os

astenman, 2020-07-27:
Hello!

In my project I needed to recompile mbed-os since I wanted to use a BLE MTU larger than the default value 23.

However, I get problems when building with the Arduino_LSM9DS1 library. It complains about Wire1 not being defined. I can fix the problem manually by using Wire instead of Wire1 in Arduino_LSM9DS1. I also have to change the definition of Wire in Wire.cpp according to

```
#if DEVICE_I2C > 0
//arduino::MbedI2C Wire(I2C_SDA, I2C_SCL);
arduino::MbedI2C Wire(I2C_SDA1, I2C_SCL1);
#endif
#if DEVICE_I2C > 1
arduino::MbedI2C Wire1(I2C_SDA1, I2C_SCL1);;
#endif
```

Am I missing something in the configuration? I am running the script "mbed-os-to-arduino" as

```
./mbed-os-to-arduino -r /home/anste/mbed-os -a ARDUINO_NANO33BLE:ARDUINO_NANO33BLE
```

where "/home/anste/mbed-os" is my local patched version of mbed os.

facchinm, 2020-07-28:
Hi @astenman ,
you are doing everything correctly, the "bug" is on our side.
`DEVICE_I2C` macro is provided by mbed build system which has no clue we are actually using 2 I2C interfaces; hence, in the updated core, we got our own macro from variant.h
https://github.com/arduino/ArduinoCore-mbed/blob/78f25290d9512857799de8a589cc4d7af362f07b/libraries/Wire/Wire.cpp#L163-L168

I suggest to switch to the new beta core for development, this one is going to be stuck to mbed 5.x forever.

astenman, 2020-07-28:
Hi @facchinm ,

Thank you for the quick response! I will try the beta core instead. Can I use the same approach for updating mbed os there?

facchinm, 2020-07-28:
Absolutely! The main pro is that you can use mbed 6 (or master version)

astenman, 2020-07-28:
@facchinm : Thank you!

closed 2020-07-28 by astenman