arduino/ArduinoCore-nRF528x-mbedos #94: Using the beta core

astenman, 2020-07-28:
Sorry to bother you again. I just tried the beta core (Arduino Mbed OS Boards / Arduino Nano 33 BLE), but I get a lot of rejected hunks while patching against the latest Mbed OS 6 master. Do I have to patch manually?

```
(pymbed) anste@Nira-PC11BYX2:/mnt/c/Users/anste/AppData/Local/Arduino15/packages/arduino-beta/hardware/mbed/1.2.1$ ./mbed-os-to-arduino -r /home/anste/mbed-os -a ARDUINO_NANO33BLE:ARDUINO_NANO33BLE

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
LOCAL_REPO=/home/anste/mbed-os
REMOTE_BRANCH=
MBED_CORE_LOCATION=/mnt/c/Users/anste/AppData/Local/Arduino15/packages/arduino-beta/hardware/mbed/1.2.1

VARIANT=ARDUINO_NANO33BLE BOARD=ARDUINO_NANO33BLE
Checking for prerequisites... done.
Creating MbedOS Application...[mbed] Working path "/tmp" (directory)
[mbed] Creating new program "mbed-os-program" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#e4b81f67f939a0c0b11c147ce74aa367271e1279"
 done.
Checking out preferred 'mbed-os' version... Linking local repo /home/anste/mbed-os... done.
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] ARDUINO_NANO33BLE now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Applying patches...can't find file to patch at input line 14
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 28dcde0a62f98bfb7a355d620cb9f6222c4f78b8 Mon Sep 17 00:00:00 2001
|From: Martino Facchin <m.facchin@arduino.cc>
|Date: Tue, 30 Jul 2019 11:47:04 +0200
|Subject: [PATCH 1/3] FlashIAPBlockDevice: add missing mbed namespace
|
|---
| .../blockdevice/COMPONENT_FLASHIAP/FlashIAPBlockDevice.h        | 2 +-
| 1 file changed, 1 insertion(+), 1 deletion(-)
|
|diff --git a/components/storage/blockdevice/COMPONENT_FLASHIAP/FlashIAPBlockDevice.h b/components/storage/blockdevice/COMPONENT_FLASHIAP/FlashIAPBlockDevice.h
|index 4cb1486dc7..5020f183a1 100644
|--- a/components/storage/blockdevice/COMPONENT_FLASHIAP/FlashIAPBlockDevice.h
|+++ b/components/storage/blockdevice/COMPONENT_FLASHIAP/FlashIAPBlockDevice.h
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
patching file drivers/source/usb/USBDevice.cpp
Hunk #1 succeeded at 980 (offset 28 lines).
patching file tools/profiles/develop.json
Hunk #1 FAILED at 5.
1 out of 1 hunk FAILED -- saving rejects to file tools/profiles/develop.json.rej
patching file tools/profiles/release.json
Hunk #1 FAILED at 5.
1 out of 1 hunk FAILED -- saving rejects to file tools/profiles/release.json.rej
patching file features/FEATURE_BLE/targets/TARGET_CORDIO/mbed_lib.json
Hunk #1 FAILED at 1.
Hunk #2 succeeded at 78 with fuzz 1 (offset 13 lines).
1 out of 2 hunks FAILED -- saving rejects to file features/FEATURE_BLE/targets/TARGET_CORDIO/mbed_lib.json.rej
 done.
Compiling Mbed Application...Compile [  0.1%]: mbed_tz_context.c
Compile [  0.2%]: AT_CellularSMS.cpp
Compile [  0.2%]: APN_db.cpp
Compile [  0.3%]: CellularLog.cpp
...
```

facchinm, 2020-07-29:
@astenman no worries :slightly_smiling_face: 
I'm going to update the patchset because it changed quite a bit during the lest dev cycle.
In the meantime you can also avoid patching anything, the nano33ble build will work out of the box since it has being mainlined in mbed.

astenman, 2020-07-29:
Thanks! But I still need to change the default BLE MTU size (23), i.e., change MBED_CONF_CORDIO_DESIRED_ATT_MTU. I thought I had to recompile mbed os in order to achieve this. But maybe I am wrong?

facchinm, 2020-07-29:
Sure, but you can call `mbed-os-to-arduino` without `-a` flag to avoid applying our patches (or use our fork https://github.com/arduino/mbed-os as a base for your modifications)

astenman, 2020-07-29:
Ah, now I see. Thanks!

closed 2020-07-29 by astenman