arduino/ArduinoCore-nRF528x-mbedos #95: Error Uploading latest edition of Nano 33 BLE

SgtGilly, 2020-08-02:
Get the following error message:

Error downloading http://downloads.arduino.cc/tools/openocd-0.10.0-arduino9-static-i686-w64-mingw32.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/tools/openocd-0.10.0-arduino9-static-i686-w64-mingw32.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/tools/openocd-0.10.0-arduino9-static-i686-w64-mingw32.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.net.SocketException: Connection reset
	at java.net.SocketInputStream.read(SocketInputStream.java:210)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at java.io.BufferedInputStream.read1(BufferedInputStream.java:284)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
	at sun.net.www.MeteredStream.read(MeteredStream.java:134)
	at java.io.FilterInputStream.read(FilterInputStream.java:133)
	at sun.net.www.protocol.http.HttpURLConnection$HttpInputStream.read(HttpURLConnection.java:3444)
	at sun.net.www.protocol.http.HttpURLConnection$HttpInputStream.read(HttpURLConnection.java:3437)
	at cc.arduino.utils.network.FileDownloader.readStreamCopyTo(FileDownloader.java:268)
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:247)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more

facchinm, 2020-08-12:
Hi @SgtGilly ,
are you still getting this error? The issue could be due to an intermittent network issue or a proxy/firewall. Can you double check if you can download the file by clicking its URL?

SgtGilly, 2020-08-23:
Yes I am still getting this error. I tried clicking the URL. It began to download a zip file but then it failed for a "Network Error".

facchinm, 2020-08-24:
It really looks like a network problem on your side. Could you test the download on the same network with another device (even a smartphone) ? At least this would allow to find the root cause.
In the meantime I'm closing the issue here since it's not really related with `mbed` core.

closed 2020-08-24 by facchinm