arduino/ArduinoCore-nRF528x-mbedos #97: Build mbed

trungnq97, 2020-08-10:
Hi @facchinm

I want to build mbed to enable `-mfloat-abi=hard`, so I run the command:

`./mbed-os-to-arduino ARDUINO_NANO33BLE:ARDUINO_NANO33BLE`

and get the error below. Do you have any idea what could be the issue? Thank you in advance.

```
MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=0
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/workspace/.platformio/packages/framework-arduino-nrf52-mbedos

VARIANT=ARDUINO_NANO33BLE BOARD=ARDUINO_NANO33BLE
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] ARDUINO_NANO33BLE now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Compiling Mbed Application...Compile [ 49.1%]: thread_management_api.c
[mbed] ERROR: "/home/trung/.platformio/penv/bin/python" returned error.
       Code: 1
       Path: "/tmp/mbed-os-program"
       Command: "/home/trung/.platformio/penv/bin/python -u /tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m ARDUINO_NANO33BLE --source . --source ./mbed-os/features/unsupported/dsp/ --build ./BUILD/ARDUINO_NANO33BLE/GCC_ARM -v"
```

facchinm, 2020-08-12:
Hi @trungnq97 ,
you can navigate to `/tmp/mbed-os-program` and try to run `mbed compile` from there to get the real error.
I think it could be due to mbed 6 changes but it's just a guess.
This core was validated with mbed 5.x so you can checkout that branch and retry running the script

trungnq97, 2020-08-12:
After upgrading my [gcc arm toolchain](https://docs.iota.org/docs/iot/0.1/nrf52/setup-guides/install-arm-gcc-toolchain#hardware) and use `mbed-os-5.15.4`. Everything works smoothly. Thank you very much, @facchinm !

closed 2020-08-12 by trungnq97