arduino/ArduinoCore-nRF528x-mbedos #99: Releases after 1.1.4 don't work

Petorrr, 2020-08-15:
My application works fine with the 1.1.4 release for the NANO BLE. Updating to 1.1.5 results in a wire compilation error. Then updating to 1.1.6 results in a failing application. The activity on the board indicates a blinking LED_BUILTIN (PIN_LED) (13), 3 slow blinks, then 3 fast blinks.
The application I run successfully with the 1.1.4 release doesn't use this LED at all, so it must be some alert or panic indication coming out of the mbed or a library like ArduinoBLE?

per1234, 2020-08-15:
The meaning of that blink pattern is explained here: https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/90#issuecomment-654062552

Petorrr, 2020-08-15:
> The meaning of that blink pattern is explained here: [#90 (comment)](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/90#issuecomment-654062552)

Thanks and I can confirm it crashes in kv_get. Any thoughts on why the use of the KvStore fails in the 1.1.6 version and not in the 1.1.4 release? I use the KvStore to store parameter values due to the lack of an EEPROM on the Nano BLE boards. KvStore seemed to offer a nice alternative to create a sort of registry of values.

manchoz, 2020-08-15:
There a few similar issues already open in upstream mbed-os. They are related to the new mandatory SecureStore component dependency.

Someone proposed a fix but I wasn't able to replicate and get it working (at least on an external SPIFBlockDevice).

We will try to track the evolving issues and post updates - and a fix - as soon as they will be available.