arduino/ArduinoCore-nRF528x-mbedos #100: 1.1.6 release:

janjongboom, 2020-08-17:
@facchinm Since the 1.1.6 release of this Arduino core we have naming issues between the Arduino and the Mbed serial interfaces. Has anything changed here?

```
Arduino: 1.8.13 (Windows 10), Board: “Arduino Nano 33 BLE”

In file included from sketch\static_buffer.ino.cpp:1:0:

C:\Users\kasim\OneDrive\Documents\Arduino\libraries\ei-plant_pot-arduino-1.0.1\examples\static_buffer\static_buffer.ino: In function ‘void setup()’:

C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:128:16: error: reference to ‘SerialUSB’ is ambiguous

#define Serial SerialUSB

            ^
C:\Users\kasim\OneDrive\Documents\Arduino\libraries\ei-plant_pot-arduino-1.0.1\examples\static_buffer\static_buffer.ino:53:5: note: in expansion of macro ‘Serial’

 Serial.begin(115200);

 ^
In file included from C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:127:0,

             from sketch\static_buffer.ino.cpp:1:
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/USB/PluggableUSBSerial.h:279:27: note: candidates are: arduino::USBSerial SerialUSB

extern arduino::USBSerial SerialUSB;

                       ^~~~~~~~~
In file included from sketch\static_buffer.ino.cpp:1:0:

C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:128:16: note: class mbed::SerialUSB

#define Serial SerialUSB

            ^
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/mbed/drivers/Serial.h:56:7: note: in expansion of macro ‘Serial’

class Serial : public SerialBase, public Stream, private NonCopyable {

   ^~~~~~
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:128:16: error: reference to ‘SerialUSB’ is ambiguous

#define Serial SerialUSB

            ^
C:\Users\kasim\OneDrive\Documents\Arduino\libraries\ei-plant_pot-arduino-1.0.1\examples\static_buffer\static_buffer.ino:55:5: note: in expansion of macro ‘Serial’

 Serial.println("Edge Impulse Inferencing Demo");

 ^
In file included from C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:127:0,

             from sketch\static_buffer.ino.cpp:1:
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/USB/PluggableUSBSerial.h:279:27: note: candidates are: arduino::USBSerial SerialUSB

extern arduino::USBSerial SerialUSB;

                       ^~~~~~~~~
In file included from sketch\static_buffer.ino.cpp:1:0:

C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:128:16: note: class mbed::SerialUSB

#define Serial SerialUSB

            ^
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/mbed/drivers/Serial.h:56:7: note: in expansion of macro ‘Serial’

class Serial : public SerialBase, public Stream, private NonCopyable {

   ^~~~~~
C:\Users\kasim\OneDrive\Documents\Arduino\libraries\ei-plant_pot-arduino-1.0.1\examples\static_buffer\static_buffer.ino: In function ‘void ei_printf(const char*, …)’:

C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:128:16: error: reference to ‘SerialUSB’ is ambiguous

#define Serial SerialUSB

            ^
C:\Users\kasim\OneDrive\Documents\Arduino\libraries\ei-plant_pot-arduino-1.0.1\examples\static_buffer\static_buffer.ino:122:9: note: in expansion of macro ‘Serial’

     Serial.write(print_buf);

     ^
In file included from C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:127:0,

             from sketch\static_buffer.ino.cpp:1:
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/USB/PluggableUSBSerial.h:279:27: note: candidates are: arduino::USBSerial SerialUSB

extern arduino::USBSerial SerialUSB;

                       ^~~~~~~~~
In file included from sketch\static_buffer.ino.cpp:1:0:

C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:128:16: note: class mbed::SerialUSB

#define Serial SerialUSB

            ^
C:\Users\kasim\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6\cores\arduino/mbed/drivers/Serial.h:56:7: note: in expansion of macro ‘Serial’

class Serial : public SerialBase, public Stream, private NonCopyable {

   ^~~~~~
exit status 1

Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
“Show verbose output during compilation”
option enabled in File -> Preferences.
```

Reported in https://forum.edgeimpulse.com/t/deployment-building-firmware-for-arduino-nano-error/625/5

Here is an Arduino library that demonstrates the problem: [ei-continuous-gestures-demo-arduino-1.0.9.zip](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/files/5083334/ei-continuous-gestures-demo-arduino-1.0.9.zip), and this is the sketch that fails to compile:

```
/* Includes ---------------------------------------------------------------- */
#include <continuous_gestures_demo_inference.h>

/**
 * @brief      Arduino setup function
 */
void setup()
{
    // put your setup code here, to run once:
    Serial.begin(115200);

    Serial.println("Edge Impulse Inferencing Demo");
}

/**
 * @brief      Arduino main function
 */
void loop()
{
    ei_printf("Edge Impulse standalone inferencing (Arduino)\n");

    delay(1000);
}

/**
 * @brief      Printf function uses vsnprintf and output using Arduino Serial
 *
 * @param[in]  format     Variable argument list
 */
void ei_printf(const char *format, ...) {
    static char print_buf[1024] = { 0 };

    va_list args;
    va_start(args, format);
    int r = vsnprintf(print_buf, sizeof(print_buf), format, args);
    va_end(args);

    if (r > 0) {
        Serial.write(print_buf);
    }
}
```

facchinm, 2020-08-17:
Hi @janjongboom ,
the issue is namespace and compilation order related.
In order to speedup the build `mbed.h` is not included anymore by default but through a wrapper that takes care to resolve ambiguities.
In `src/edge-impulse-sdk/dsp/numpy_types.h"` there's a `using namespace mbed;` directive that gets applied to all the compilation units.
I understand that specifying the namespace of every Callback call is tedious but since all the library is implemented in header files and they include each other every `using namespace ...`  gets applied to the final compilation unit.

I can provide a patch is needed but I'd avoid changing the core internals (again)

janjongboom, 2020-08-19:
@facchinm I can get rid of our `using namespace mbed` calls, but I bet there are heaps of other libraries that have the same behavior, that will break now in a new minor version of the core.